open62541/open62541 #2880: Server cannot create instance of GeneralModelChangeEventType

inspektor-scherlein, 2019-07-11:
## Description
My open62541-based server creates and removes nodes in the server AddressSpace at runtime. To notify the client side, I'd like to use the GeneralModelChangeEventType, but the creation of an event of that type fails. I'm using UaExpert V.1.5.1 on the client side.

## Background Information / Reproduction Steps
Trying to create the event using UA_Server_createEvent fails because GeneralModelChangeEventType is abstract. I'd like to point out that the standard says, you are allowed to instantiate abstract EventTypes, as long as they don't appear in the AddressSpace. See OPC UA Part 3, chapter 4.6.2 EventTypes:
> EventTypes defined in this document are specified as abstract and therefore never instantiated in the AddressSpace. Event occurrences of those EventTypes are only exposed via a Subscription. EventTypes exist in the AddressSpace to allow Clients to discover the EventType. This information is used by a client when establishing and working with Event Subscriptions. EventTypes defined by other parts of this series of standards or companion specifications as well as Server specific EventTypes may be defined as not abstract and therefore instances of those EventTypes may be visible in the AddressSpace although Events of those EventTypes are also accessible via the Event Notification mechanisms.

Because UA_Server_createEvent tries to create a node with no parent node, I think this requirement is fulfilled.

Another reference going in that direction can be found in chapter 9.32.7 Guidelines for ModelChangeEvents:
>Two types of ModelChangeEvents are defined: the BaseModelChangeEvent that does not contain any information about the changes and the GeneralModelChangeEvent that identifies the changed Nodes via an array. The precision used depends on both the capability of the OPC UA Server and the nature of the update. An OPC UA Server may use either ModelChangeEvent type depending on circumstances. It may also define subtypes of these EventTypes adding additional information.

In other words: inheriting from GeneralModelChangeEvent is optional, I may also use GeneralModelChangeEvent directly.

I also tried this with the Unified Automation ANSI C Demo Server, which provides methods for dynamic node creation/deletion in its address space at Root\Objects\Demo\0008_DynamicNodes. Those methods instantiate GeneralModelChangeEvents directly.

So IMHO this is a bug, do you agree? Or is there a different way to create instances of those abstract event types?

Used CMake options:

```bash
cmake -DUA_ENABLE_AMALGAMATION=OFF -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON
```
Code snippet:
```
UA_NodeId eventID;

// mServer points to a UA_Server object
UA_StatusCode result = UA_Server_createEvent(mServer, UA_NODEID_NUMERIC(0, UA_NS0ID_GENERALMODELCHANGEEVENTTYPE), &eventID);

// UA_STATUSCODE_BADTYPEDEFINITIONINVALID is returned
```

## Checklist
 - [x] open62541 Version: commit 21e32dd9973096133a2fcf826809518e5fd0bdfc
 - [x] Other OPC UA SDKs used (client or server): UaExpert V1.5.1
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-07-11:
Agree. It should be possible to create events using abstract event types.
@aDogCalledSpot can you have a look?

closed 2019-07-25 by Pro

juangburgos, 2019-08-13:
I saw this was merged to RC1, is the fix also available in master?

Pro, 2019-08-14:
Yes, that fix shoul be in both braches.

juangburgos, 2019-08-19:
@Pro thanks. 

@inspektor-scherlein did you manage to get UaExplorer to refresh itself by triggering a `UA_NS0ID_GENERALMODELCHANGEEVENTTYPE`? I have been trying different ways without success.

If so, could you please post a code snippet on how you did it? Thank you in advance.

inspektor-scherlein, 2019-10-14:
Hi @juangburgos,
sorry for my late response. I must have missed the notification e-mail about your post...
But yes, I got this working with UaExpert. For example, if you add a node and want to notify about that, you also need to include the parent node that has a reference to the new node into the notification:

```
UA_NodeID eventId;
UA_Server_createEvent(server, UA_NODEID_NUMERIC(0, UA_NS0ID_GENERALMODELCHANGEEVENTTYPE), &eventId);

// set event time
UA_DateTime eventTime = UA_DateTime_now();
UA_Server_writeObjectProperty_scalar(server, eventId, UA_QUALIFIEDNAME(0, "Time"), &eventTime, &UA_TYPES[UA_TYPES_DATETIME]);

// set event severity
UA_UInt16 eventSeverity = 100;
UA_Server_writeObjectProperty_scalar(server, eventId, UA_QUALIFIEDNAME(0, "Severity"), &eventSeverity, &UA_TYPES[UA_TYPES_UINT16]);

UA_ModelChangeStructureDataType changes[2];

// notify about adding node "newNodeId" with type "newNodeIdType"
UA_ModelChangeStructureDataType_init(&changes[0]);
changes[0].affected = newNodeId;
changes[0].affectedType = newNodeIdType;
changes[0].verb = UA_MODELCHANGESTRUCTUREVERBMASK_NODEADDED;

// notify about new parent node reference for "parentNodeId" with "parentNodeIdType"
UA_ModelChangeStructureDataType_init(&changes[1]);
changes[1].affected = parentNodeId;
changes[1].affectedType = parentNodeIdType;
changes[1].verb = UA_MODELCHANGESTRUCTUREVERBMASK_REFERENCEADDED;

// write Changes property
UA_Variant changesProperty;
UA_Variant_setArrayCopy(&changesProperty, changes, 2, &UA_TYPES[UA_TYPES_MODELCHANGESTRUCTUREDATATYPE]);

UA_Server_writeObjectProperty(server, eventId, UA_QUALIFIEDNAME(0, "Changes"), changesProperty);

// cleanup variant and exit, if failed => left out here

UA_ServerTriggerEvent(server, eventId, UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER), NULL, UA_TRUE);

UA_Variant_clear(&changesProperty);

```

juangburgos, 2019-10-14:
@inspektor-scherlein , thanks a lot, after some soffering I managed too. The snippet will help me check if I can improve. Thanks again.

PR #2889 (2019-07-25): Allow abstract events
Fixes #2880