open62541/open62541 #2882: Build on Alpine Linux Broken

marcotuna, 2019-07-11:
## Description

Using the latest version (master) is not possible to build for Alpine Linux. On previous versions it was building perfectly. For now only Alpine Linux version 3.9 is known to not work. Using Debian or Ubuntu works just fine.

## Background Information / Reproduction Steps

- Launch an Alpine Linux VM or a Docker Container (with Alpine Linux).
- Install dependencies

```bash
RUN apk update && apk upgrade && apk add --no-cache \
bash \ 
git \
cmake \
gcc \
g++ \
musl-dev \
python \
py-pip \
make \
jansson-dev \
sqlite-dev \
mbedtls-dev
```

- Build  the project

## Logs

```
Current branch master is up to date.
[ 16%] Performing configure step for 'EP_open62541'
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.16") 
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Architectures included in amalgamation: win32;posix
-- The selected architecture is: posix
-- Could NOT find MBEDTLS (missing: MBEDTLS_INCLUDE_DIRS MBEDTLS_LIBRARY MBEDX509_LIBRARY MBEDCRYPTO_LIBRARY) 
-- Performing Test CC_HAS_-fstack-protector-strong
-- Performing Test CC_HAS_-fstack-protector-strong - Success
-- Performing Test CC_HAS_-fstack-clash-protection
-- Performing Test CC_HAS_-fstack-clash-protection - Success
-- Performing Test CC_HAS_-mcet -fcf-protection
-- Performing Test CC_HAS_-mcet -fcf-protection - Failed
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/home/user/TestProject/depends/open62541/src/EP_open62541/MBEDTLS_INCLUDE_DIRS
   used as include directory in directory /home/user/TestProject/depends/open62541/src/EP_open62541
MBEDCRYPTO_LIBRARY (ADVANCED)
    linked by target "open62541" in directory /home/user/TestProject/depends/open62541/src/EP_open62541
MBEDTLS_LIBRARY (ADVANCED)
    linked by target "open62541" in directory /home/user/TestProject/depends/open62541/src/EP_open62541
MBEDX509_LIBRARY (ADVANCED)
    linked by target "open62541" in directory /home/user/TestProject/depends/open62541/src/EP_open62541

-- Configuring incomplete, errors occurred!
See also "/home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeOutput.log".
See also "/home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeError.log".
make[2]: *** [src/CMakeFiles/EP_open62541.dir/build.make:106: ../depends/open62541/src/EP_open62541-stamp/EP_open62541-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:95: src/CMakeFiles/EP_open62541.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

```
~/TestProject # cat /home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeError.log
Performing C SOURCE FILE Test CC_HAS_-mcet -fcf-protection failed with the following output:
Change Dir: /home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_a4ee3/fast"
make[3]: Entering directory '/home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_a4ee3.dir/build.make CMakeFiles/cmTC_a4ee3.dir/build
make[4]: Entering directory '/home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_a4ee3.dir/src.c.o
/usr/bin/cc   -DCC_HAS_-mcet -fcf-protection   -mcet -fcf-protection -o CMakeFiles/cmTC_a4ee3.dir/src.c.o   -c /home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeTmp/src.c
cc: error: unrecognized command line option '-mcet'; did you mean '-maes'?
make[4]: *** [CMakeFiles/cmTC_a4ee3.dir/build.make:66: CMakeFiles/cmTC_a4ee3.dir/src.c.o] Error 1
make[4]: Leaving directory '/home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeTmp'
make[3]: *** [Makefile:121: cmTC_a4ee3/fast] Error 2
make[3]: Leaving directory '/home/user/TestProject/depends/open62541/src/EP_open62541-build/CMakeFiles/CMakeTmp'

Source file was:
int main(void) { return 0; }
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master (latest)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Alpine Linux 3.9
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-07-12:
Did you manually enable encryption?

Then, mbedtls is required as an additional dependency.

marcotuna, 2019-07-12:
mbedtls-dev is now installed. Now I get the following error:

```
[ 83%] Building C object src/CMakeFiles/TestProject.dir/modules/opcua/opcua.c.o
[ 86%] Building C object src/CMakeFiles/TestProject.dir/modules/opcua/convert.c.o
[ 90%] Building C object src/CMakeFiles/TestProject.dir/modules/opcua/browse.c.o
[ 93%] Building C object src/CMakeFiles/TestProject.dir/modules/opcua/read.c.o
[ 96%] Building C object src/CMakeFiles/TestProject.dir/modules/opcua/monitor.c.o
make[2]: *** No rule to make target '../depends/open62541/lib/libopen62541.a', needed by 'src/TestProject'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:169: src/CMakeFiles/TestProject.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

marcotuna, 2019-07-12:
I performed a find for the file libopen62541.a in both systems and found out that in Alpine Linux is writting to the folder lib64 

Alpine 3.9:
```
./depends/open62541/lib64/libopen62541.a
./depends/open62541/src/EP_open62541-build/bin/libopen62541.a
```
Debian Stretch
```
./depends/open62541/lib/libopen62541.a
./depends/open62541/src/EP_open62541-build/bin/libopen62541.a
```

Pro, 2019-07-12:
Did you already see our "official" Dockerfile?
https://github.com/open62541/open62541/blob/master/Dockerfile

It's currently not really usable, but with this we test building in our CI.
I will extend that a bit to make it more usable, i.e., use make install & Co.

This issue with the lib folders may have been fixed a few days ago with #2851

Are you using the current master branch for your tests?

marcotuna, 2019-07-12:
I managed to change the include directories in my CMake and is now working. I am closing this issue.

closed 2019-07-12 by marcotuna