open62541/open62541 #2887: create freertos amalgamation files

mrburen121, 2019-07-15:
## Description
I get a error when using cmake to create the freertos amalgamation files.

## Background Information / Reproduction Steps
Used CMake options:
`cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ARCHITECTURE=freertosLWIP -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_PUBSUB=ON -DUA_ARCH_FREERTOS_USE_OWN_MEMORY_FUNCTIONS=ON ..`

This is the output:
```
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3.7 (found version "3.7.3") 
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Architectures included in amalgamation: freertosLWIP
-- The selected architecture is: freertosLWIP
-- Performing Test CC_HAS_-fstack-protector-strong
-- Performing Test CC_HAS_-fstack-protector-strong - Success
-- Performing Test CC_HAS_-fstack-clash-protection
-- Performing Test CC_HAS_-fstack-clash-protection - Success
-- Performing Test CC_HAS_-mcet -fcf-protection
-- Performing Test CC_HAS_-mcet -fcf-protection - Failed
-- Performing Test CC_HAS_-D_FORTIFY_SOURCE=2
-- Performing Test CC_HAS_-D_FORTIFY_SOURCE=2 - Success
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
-- Configuring done
-- Generating done
```

While running the make command I get this error:
```
Scanning dependencies of target open62541-object
[ 65%] Building C object CMakeFiles/open62541-object.dir/open62541.c.o
In file included from /home/user/GitHub/open62541/build/open62541.c:28:
/home/user/GitHub/open62541/build/open62541.h:237:10: fatal error: lwip/tcpip.h: No such file or directory
 #include <lwip/tcpip.h>
          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/open62541-object.dir/build.make:266: CMakeFiles/open62541-object.dir/open62541.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:337: CMakeFiles/open62541-object.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```
Where can I find the missing file(s)?

Kind regards,

Chris

Please provide the following information:

 - [X] open62541 Version (release number or git tag): current version
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Ubuntu 19.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-07-15:
@cabralfortiss ping?

cabralfortiss, 2019-07-16:
pong! you need to set the variable UA_ARCH_EXTRA_INCLUDES with the folder or folders where your lwip header files are.

Check https://github.com/open62541/open62541/blob/master/arch/CMakeLists.txt to see what can you adapt for your compilation

mrburen121, 2019-07-16:
Thank you. I included my lwip folder where the the includes of ua_lwip.h are located.
```
#include <lwip/tcpip.h>
#include <lwip/netdb.h>
#include <lwip/init.h>
#include <lwip/sockets.h>
```
But, and outsource, these files have other dependencies like lwipopts.h and my lpc board files.
Is there a sample project / lwip folder available that I can use as a reference? Or could you suggest me how to create the files with my folder structure, my repo: [https://bitbucket.org/mrburen121/opcua_server_freertos/src/master/OPCUA-RTOSServer/](https://bitbucket.org/mrburen121/opcua_server_freertos/src/master/OPCUA-RTOSServer/)

cabralfortiss, 2019-07-16:
You can add many folder paths divided by a semicolon. You need to add all of them. I did it by adding one, checking where was the next missing folder, adding that and so on until it compiled

mrburen121, 2019-07-16:
Thank you. When I run the command
`cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ARCHITECTURE=freertosLWIP -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_PUBSUB=ON -DUA_ARCH_FREERTOS_USE_OWN_MEMORY_FUNCTIONS=ON -DUA_ARCH_EXTRA_INCLUDES=/home/user/opcua/OPCUA-RTOS-Server/OPCUA-RTOSServer/inc;/home/user/opcua/OPCUA-RTOS-Server/OPCUA-RTOSServer/lwip/include ..`
I get this error:
```
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/home/chris/GitHub/open62541/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
bash: /home/user/opcua/OPCUA-RTOS-Server/OPCUA-RTOSServer/lwip: Is a directory
```
Running the command without the second string works perfect. Am I doing something wrong?

cabralfortiss, 2019-07-16:
Not sure, I do it using the GUI of CMake. I'd say to try it with double quotes:

-DUA_ARCH_EXTRA_INCLUDES="/home/user/opcua/OPCUA-RTOS-Server/OPCUA-RTOSServer/inc;/home/user/opcua/OPCUA-RTOS-Server/OPCUA-RTOSServer/lwip/include"

mrburen121, 2019-07-16:
Thank you, this solved my problems. I get a lot of compile errors from my lpc stack. I hope to solve them soon.
Did you got this error? How can I solve this?
`error: ISO C99 doesn’t support unnamed structs/unions [-Werror=pedantic]`

cabralfortiss, 2019-07-16:
https://github.com/open62541/open62541/blob/master/arch/CMakeLists.txt#L16

Pro, 2019-07-16:
I will close this issue since it seems to be solved. @mrburen121 please also check the other issues related to FreeRTOS, which contain a lot of viable information to fix your build issues, especially #2511

closed 2019-07-16 by Pro

Gaurav7611, 2021-01-14:
Hi Experts
Hope you are doing good
I am facing weird problem while generating open62541.c/.h files.
in build folder inside open62541 I run below command to generate open62541.c/.h files
"cmake -DUA_ARCHITECTURE=freertosLWIP -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_PUBSUB=ON.."
I see that the cmake throws an error as below :
Make Error: The source directory "/home/gaurav/tmp/open62541/build_freertos" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

I am new to this, Please help me out regarding this.

Thanks

cochicde, 2021-01-14:
you're missing a space before the two dots at the end, after PUBSUB=ON

-DUA_ENABLE_PUBSUB=ON ..

instead of 

-DUA_ENABLE_PUBSUB=ON..