open62541/open62541 #2890: Event-based Subscriber OPCUA pubsub

canadyre, 2019-07-16:
## Description
Hi all,
I am trying to run the pubsub examples on Raspberry Pis. Right now, the subscriber is using a polling mechanism set to some number of milliseconds. Is there a way to change this to event-based or are there any examples where this is done?
Thanks!

## Background Information / Reproduction Steps

Used CMake options: 

UA_ENABLE_PUBSUB ON
UA_ENABLE_PUBSUB_ETH_UADP ON
UA_BUILD_EXAMPLES ON
CMAKE_C_COMPILER /usr/bin/arm-linux-gnueabihf-gcc

## Checklist

Please provide the following information:

 - [0.3.0] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [Raspbian w/ RT-Preempt patch ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

sohailmushtaq, 2019-07-16:
Hi @canadyre,
what do you mean with " the subscriber is using a polling mechanism set to some number of milliseconds".
Are you using UDP?

canadyre, 2019-07-16:
Hi @sohailmushtaq ,
Yes I am using UDP right, and I just mean that the subscriber polls for a message every 5-10 ms.

sohailmushtaq, 2019-07-16:
Ok, @canadyre.
If you used MQTT you saw that they use a callback for receiving messages and with UDP there is no such callback mechanism in the code. I think for now the polling remains because they need to do polling on the channel where they receive. Now I don't remember the details of the code but inside I saw that you have to do this polling to receive information at Subscriber side. I hope I'm understandable :)

sohailmushtaq, 2019-07-16:
Hi @canadyre, 
I have a question for pubsub over Rasberry Pi. Does the library work well? Do you try UDP and MQTT on Rasberry? 
Thanks!

canadyre, 2019-07-16:
Hi @sohailmushtaq , 
Thanks for the help with that! I've used MQTT so I'll look into that some more.

As for your question, it has worked pretty well as far as I can tell. You just have to use a cross compiler, I usually compile on my desktop and then send to the Raspberry Pi. Also, you have to modify the code to work with 32 bits vs 64, but otherwise it has worked well.

sohailmushtaq, 2019-07-16:
Thanks @canadyre!
With " you have to modify the code to work with 32 bits vs 64" you mean that you can't use Double and everything that's based on 64 bits? 
Thanks!