open62541/open62541 #2893: Compile freeRTOS with open62541 errors

mrburen121, 2019-07-18:
## Description
After creating the amalgamation header/source files and included them in my project I get some compiling errors. Did I forgot to include some parameters/files?

Language: -std=gnu99 (-std=c99 is also not working)
Compiler flags: -c -fmessage-length=0 -fno-builtin -ffunction-sections -fdata-sections -UA_ARCHITECTURE_FREERTOSLWIP

```
In file included from ../src/open62541.c:28:
../src/open62541.c: In function 'UA_PubSubChannelUDPMC_open':
/home/chris/GitHub/OPCUA-RTOSServer/lwip_freertos_tcpecho/inc/open62541.h:284:27: warning: implicit declaration of function 'ip4addr_aton'; did you mean 'ipaddr_aton'? [-Wimplicit-function-declaration]
      (((af) == AF_INET) ? ip4addr_aton((src),(ip4_addr_t*)(dst)) : 0)
                           ^~~~~~~~~~~~
../src/open62541.c:56076:9: note: in expansion of macro 'UA_inet_pton'
         UA_inet_pton(AF_INET, addressAsChar, &imr_interface);
         ^~~~~~~~~~~~
/home/chris/GitHub/OPCUA-RTOSServer/lwip_freertos_tcpecho/inc/open62541.h:284:47: error: 'ip4_addr_t' undeclared (first use in this function); did you mean 'ip_addr_t'?
      (((af) == AF_INET) ? ip4addr_aton((src),(ip4_addr_t*)(dst)) : 0)
                                               ^~~~~~~~~~
../src/open62541.c:56076:9: note: in expansion of macro 'UA_inet_pton'
         UA_inet_pton(AF_INET, addressAsChar, &imr_interface);
         ^~~~~~~~~~~~
/home/chris/GitHub/OPCUA-RTOSServer/lwip_freertos_tcpecho/inc/open62541.h:284:47: note: each undeclared identifier is reported only once for each function it appears in
      (((af) == AF_INET) ? ip4addr_aton((src),(ip4_addr_t*)(dst)) : 0)
                                               ^~~~~~~~~~
../src/open62541.c:56076:9: note: in expansion of macro 'UA_inet_pton'
         UA_inet_pton(AF_INET, addressAsChar, &imr_interface);
         ^~~~~~~~~~~~
/home/chris/GitHub/OPCUA-RTOSServer/lwip_freertos_tcpecho/inc/open62541.h:284:58: error: expected expression before ')' token
      (((af) == AF_INET) ? ip4addr_aton((src),(ip4_addr_t*)(dst)) : 0)
                                                          ^
../src/open62541.c:56076:9: note: in expansion of macro 'UA_inet_pton'
         UA_inet_pton(AF_INET, addressAsChar, &imr_interface);
         ^~~~~~~~~~~~
../src/open62541.c:56100:81: error: invalid application of 'sizeof' to incomplete type 'struct sockaddr_storage'
     channelDataUDPMC->ai_addr = (struct sockaddr_storage *) UA_calloc(1, sizeof(struct sockaddr_storage));
                                                                                 ^~~~~~
../src/open62541.c:56123:22: error: 'IP_MULTICAST_LOOP' undeclared (first use in this function); did you mean 'IP_MULTICAST'?
                      IP_MULTICAST_LOOP,
                      ^~~~~~~~~~~~~~~~~
                      IP_MULTICAST
../src/open62541.c:56145:22: error: 'IP_MULTICAST_TTL' undeclared (first use in this function); did you mean 'IP_MULTICAST'?
                      IP_MULTICAST_TTL,
                      ^~~~~~~~~~~~~~~~
                      IP_MULTICAST
../src/open62541.c:56178:28: error: field 'ipv4' has incomplete type
             struct ip_mreq ipv4;
                            ^~~~
In file included from ../src/open62541.c:28:
/home/chris/GitHub/OPCUA-RTOSServer/lwip_freertos_tcpecho/inc/open62541.h:284:58: error: expected expression before ')' token
      (((af) == AF_INET) ? ip4addr_aton((src),(ip4_addr_t*)(dst)) : 0)
                                                          ^
../src/open62541.c:56183:12: note: in expansion of macro 'UA_inet_pton'
         if(UA_inet_pton(AF_INET, interfaceAsChar, &group.ipv4.imr_interface)){
            ^~~~~~~~~~~~
../src/open62541.c:56203:28: error: 'IP_MULTICAST_IF' undeclared (first use in this function); did you mean 'IP_MULTICAST'?
                            IP_MULTICAST_IF,
                            ^~~~~~~~~~~~~~~
                            IP_MULTICAST
../src/open62541.c:56182:11: warning: variable 'group' set but not used [-Wunused-but-set-variable]
         } group;
           ^~~~~
../src/open62541.c: In function 'UA_PubSubChannelUDPMC_regist':
../src/open62541.c:56238:24: error: storage size of 'groupV4' isn't known
         struct ip_mreq groupV4;
                        ^~~~~~~
../src/open62541.c:56239:115: error: invalid application of 'sizeof' to incomplete type 'struct ip_mreq'
         memcpy(&groupV4.imr_multiaddr, &((const struct sockaddr_in *)connectionConfig->ai_addr)->sin_addr, sizeof(struct ip_mreq));
                                                                                                                   ^~~~~~
../src/open62541.c:56243:55: error: 'IP_ADD_MEMBERSHIP' undeclared (first use in this function)
         if(UA_setsockopt(channel->sockfd, IPPROTO_IP, IP_ADD_MEMBERSHIP, (char *) &groupV4, sizeof(groupV4)) != 0) {
[... 87 lines cut ...]
In FreeRTOSConfig.h I set:
#define configCHECK_FOR_STACK_OVERFLOW 1
#define configUSE_MALLOC_FAILED_HOOK 1

In my main.c I only included the open62541 header file. I did not add any open62541 server code.

In open62541.h the following files are included, but AI_PASSIVE isn't declared in one of those files, even not i any file in my project.
```
#include "lwip/tcpip.h"
#include "lwip/netdb.h"
#include "lwip/init.h"
#include "lwip/sockets.h"
```

## Background Information / Reproduction Steps

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 0.4.0 dev  (last GitHub version)
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Ubuntu 19.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

cabralfortiss, 2019-07-18:
You said you use the flag -UA_ARCHITECTURE_FREERTOSLWIP (not sure where) but this might be -DUA_ARCHITECTURE_FREERTOSLWIP instead. In most IDE, you can add symbols to your project, and then it generates the flag according to it.

If your generating the amalgamted files and including them in a project in an IDE prepared for your embedded, then when you do cmake you only need -DUA_ARCHITECTURE=freertosLWIP regarding the architecture and the compilation will fail, but the generated files will be there. The other flags are needed when compiling, which should be set by the IDE

cabralfortiss, 2019-07-18:
pub sub wasn't tested yet. Try deactivating. Again, there isn't much we can do from here. If AI_PASSIVE is not recognized, you'll have to look for it in your code. The information provided in #2511 has made most people sucessfully run the opcua server. If something is working, probably you did something different. Take your time and read it carefully and investigate where the problem is coming from. Your problems seem to be related to your project and not to the open62541 code

mrburen121, 2019-07-18:
I solved the errors. Thank you for your help. I added some missing parameters to my header file.

When I add #2511 the functions:
```
void vApplicationMallocFailedHook(){
	for(;;){
		vTaskDelay(1000);
	}
}

void vApplicationStackOverflowHook( TaskHandle_t xTask, char *pcTaskName ){
	for(;;){
		vTaskDelay(1000);
	}
}
```
I get the error: undefined reference to `pdMS_TO_TICKS'. Does this mean I have some timing issues? 

Also in the open62541.h file that I generated via cmake there is no function called
`UA_ServerConfig_new_customBuffer`. Is this changed in the newest version to UA_ServerConfig_setMinimalCustomBuffer?

cabralfortiss, 2019-07-18:
I have pdMS_TO_TICKS defined in freertos/include/projdefs.h 
It should be defined in your freertos somewhere

About UA_ServerConfig_new_customBuffer can maybe @Pro help

Pro, 2019-07-19:
> ... there is no function called
> `UA_ServerConfig_new_customBuffer`. Is this changed in the newest version to UA_ServerConfig_setMinimalCustomBuffer?

Yes correct, we changed how the server config is handled. You can find more on that in our examples on master.

mrburen121, 2019-07-22:
Thank you. I updated my FreeRTOS files to V10.2.0 and this solved the missing pdMS_TO_TICKS problem.

For now I have two questions left:
1. What heap file are you using? Heap 1,2,3,4 or 5?
2. Below my server code. Is this correct? If yes then I'm going to find the problem in my micro controller (driver) files.

```
/* Sets up system hardware */
void prvSetupHardware(void)
{
	/* LED0 is used for the link status, on = PHY cable detected */
	SystemCoreClockUpdate();
	Board_Init();

	/* Initial LED state is off to show an unconnected cable state */
	for (int i = 0; i < 8 ; i++) {
		Board_LED_Set(i, false);
	}
}

void vApplicationMallocFailedHook(){
	for(;;){
		vTaskDelay(pdMS_TO_TICKS(1000));
		DEBUGOUT("vApplicationMallocFailedHook");
	}
}

void vApplicationStackOverflowHook( TaskHandle_t xTask, char *pcTaskName ){
	for(;;){
		vTaskDelay(pdMS_TO_TICKS(1000));
		DEBUGOUT("vApplicationStackOverflowHook");
	}
}

static void opcua_thread(void *arg)
{
   //Allows to set smaller buffer for the connections, which can cause problems
	UA_UInt32 sendBufferSize = 8000; //64 KB was too much for my platform
	UA_UInt32 recvBufferSize = 8000;  //64 KB was too much for my platform

	UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);

	UA_ServerConfig_setMinimalCustomBuffer(config, 4840, 0, sendBufferSize, recvBufferSize);

	//VERY IMPORTANT: Set the hostname with your IP before allocating the server
	UA_ServerConfig_setCustomHostname(config, UA_STRING("192.168.1.10"));
	UA_ServerConfig_setDefault(UA_Server_getConfig(server));

	//example
	// https://open62541.org/doc/current/tutorial_server_variable.html

	UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the.answer");

	/* Write a different integer value */
	UA_Int32 myInteger = 43;
	UA_Variant myVar;
	UA_Variant_init(&myVar);
	UA_Variant_setScalar(&myVar, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
	UA_Server_writeValue(server, myIntegerNodeId, myVar);

	/* Set the status code of the value to an error code. The function
	 * UA_Server_write provides access to the raw service. The above
	 * UA_Server_writeValue is syntactic sugar for writing a specific node
	 * attribute with the write service. */
	UA_WriteValue wv;
	UA_WriteValue_init(&wv);
	wv.nodeId = myIntegerNodeId;
	wv.attributeId = UA_ATTRIBUTEID_VALUE;
	wv.value.status = UA_STATUSCODE_BADNOTCONNECTED;
	wv.value.hasStatus = true;
	UA_Server_write(server, &wv);

	/* Reset the variable to a good statuscode with a value */
	wv.value.hasStatus = false;
	wv.value.value = myVar;
	wv.value.hasValue = true;
	UA_Server_write(server, &wv);
}

int main(void) {
//	signal(SIGINT, stopHandler);
//	signal(SIGTERM, stopHandler);

	prvSetupHardware();

	/* Add another thread for initializing physical interface. This
	   is delayed from the main LWIP initialization. */
	xTaskCreate(vSetupIFTask, "SetupIFx",
				configMINIMAL_STACK_SIZE, NULL, (tskIDLE_PRIORITY + 1UL),
				(xTaskHandle *) NULL);
	/* LED1 toggle thread */
	xTaskCreate(vLEDTask1, "vTaskLed1", configMINIMAL_STACK_SIZE,
			NULL, (tskIDLE_PRIORITY + 1UL), (TaskHandle_t *) NULL);

	/* LED2 toggle thread */
	xTaskCreate(vLEDTask2, "vTaskLed2", configMINIMAL_STACK_SIZE,
			NULL, (tskIDLE_PRIORITY + 1UL), (TaskHandle_t *) NULL);

	/* UART output thread, simply counts seconds */
	xTaskCreate(vUARTTask, "vTaskUart", configMINIMAL_STACK_SIZE,
			NULL, (tskIDLE_PRIORITY + 1UL), (TaskHandle_t *) NULL);

	// Start opcua server
	printf("Start opcua server");
	if(sys_thread_new("opcua_thread", opcua_thread, NULL, 8000, 3) == NULL)
		 DEBUGOUT("opcua(): Task creation failed.", 0);

	/* Start the scheduler */
	vTaskStartScheduler();

	/* Should never arrive here */
	return 1;
}
```
With the lines below commented so is it not a part of the program, the LED's are blinking and the internet connection is able to be established.
```
if(sys_thread_new("opcua_thread", opcua_thread, NULL, 8000, 5) == NULL)
	 DEBUGOUT("opcua(): Task creation failed.", 0);
```
It never come to the line with the debugout print.

cabralfortiss, 2019-07-22:
> 
> For now I have two questions left:
> 
>     1. What heap file are you using? Heap 1,2,3,4 or 5?

I'm using 3 with realloc and calloc self-implemented just calling pvPortMalloc

> static void opcua_thread(void *arg)
> {
>    //Allows to set smaller buffer for the connections, which can cause problems
> 	UA_UInt32 sendBufferSize = 8000; //64 KB was too much for my platform [...]

I think this last line shouldn't be there since it will reset the hostname you set the line before

> //example
> 	// https://open62541.org/doc/current/tutorial_server_variable.html
>
> 	UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the.answer");
>
> 	/* Write a different integer value */
> 	UA_Int32 [...]

You're not starting your server. 

From the example, you see at the end of the function the following:

 UA_StatusCode retval = UA_Server_run(server, &running); //Server runs here until the boolean running is set to false from someone else
 UA_Server_delete(server);

mrburen121, 2019-07-22:
Oke, thank you. It seems to be a memory problem.
[image]

Could this be solved when I implement realloc and calloc? I know this is not a open62541 problem but I hope you can help me. Do you have a sample of those implementations? I can't find one for heap 3 the freertos V10 demo folder.

svro86, 2019-11-11:
Hey guys, I am new to github so apologies if I am in the wrong thread. I am trying to set up open62541 as a opc server on stm F207. The tutorial for creating open62541.c and open62541.h works fine but after integrating these files into my stmCubeIDE Project i have about 30 compile errors left. I am pretty sure that I miss an include or something and maybe someone could give me a hint. 

I am working with the latest master-branch, maybe this is my error? Thanks in advance for any help.

The Pictures shows the compile errors I have left:
[image]

cabralfortiss, 2019-11-11:
Read the documentation carefully.

It looks like you forgot this part:

When compiling LwIP you need a file called lwipopts.h. In this file, you put all the configuration variables. You need to make sure that you have the following configurations there:

```
#define LWIP_COMPAT_SOCKETS 0 // Don't do name define-transformation in networking function names.
#define LWIP_SOCKET 1 // Enable Socket API (normally already set)
#define LWIP_DNS 1 // enable the lwip_getaddrinfo function, struct addrinfo and more.
#define SO_REUSE 1 // Allows to set the socket as reusable
#define LWIP_TIMEVAL_PRIVATE 0 // This is optional. Set this flag if you get a compilation error about redefinition of struct timeval
```

svro86, 2019-11-11:
@cabralfortiss Thanks for the quick answer. I have these defines in my lwipopts.h and double checked it now. I did the changes of the freertos-config file as well. Before i had them I had around 300 errors ;)

I´m going to try the release pack v1.0 later or tomorrow instead of the masterbranch I use now. Or is their another stable release of open62541 I haven´t seen yet.

cabralfortiss, 2019-11-11:
mmm that's strange, all the errors are related to the lwip, and those flags should take care of it. Maybe check that these values are actually there when used in open62541.

In you open62541.h after the includes like #include <lwip/sockets.h> put some #if to check that for example LWIP_COMPAT_SOCKETS  is actually 0

svro86, 2019-11-12:
@cabralfortiss I actually don´t know what went wrong yesterday. Overnight I thought there might be a problem because I use RMII ethernet config insted of MII but that was not the problem. I did set up a whole new project in another workspace this morning and now I can compile it with some other errors which I am trying to fix at the moment.

1) From the [example](https://open62541.org/doc/current/building.html#freertos-lwip) opcua_thread:
for the line 
UA_ServerConfig_setMinimal(uaServerConfig, portNumber, 0, sendBufferSize, recvBufferSize)
my compiler said that the function has too many arguments and after looking I took the buffers out because they are not used.

2) in lwipopts.h I had to change the default value of MEMP_NUM_SYS_TIMEOUT to something higher than the default value. I couldn´t try yet what will happen with the whole application because of this but it shouldn´t be much of a problem.

3) I have problems with the freeRTOSConfig.h flags (they are for debugging?) because of using heap_4.c. I just set them to 0 to see if I can compile now and it works. I have seen possible solutions for this problem in the other threads mainly in #2511

ZekeLandon, 2020-02-04:
@mrburen121 hey Chris, did you manage to run PubSub in your embedded device?

mrburen121, 2020-02-07:
Hello @ZekeLandon. No, I didn't manage to get it working. The reason for this is that we currently use the TCPIP stack and the open62541 project the LWIP stack. It would cost us to much time to add TCPIP as architecture to open62541. It has currently not the highest priority to implement OPC UA in our products. My guess is that I will start on the implenentation of TCPIP in Q3 this year.

Pro, 2020-02-12:
Maybe this example helps:

https://github.com/Pro/open62541-esp32

It shows how to compile open62541 for the ESP32 microchip, which is based on FreeRTOS.

mohamedalielghoula, 2020-07-23:
I got this error during the compilation :
../Core/Src/open62541.c:56486:46: error: dereferencing pointer to incomplete type 'struct sockaddr_in6'

teccheck, 2025-02-25:
This issue is obsolete as we have a completely new event loop now. However, work on FreeRTOS is still ongoing

closed 2025-02-25 by teccheck