open62541/open62541 #2898: UA_NodeId_order incorrectly compare UA_NodeId

melkorb, 2019-07-19:
## Description
UA_NodeId_order incorrectly compare UA_NodeIds when UA_NodeId.identifierType equal UA_NODEIDTYPE_GUID.
## Background Information / Reproduction Steps
**current code: ua_types.c line 311**
case UA_NODEIDTYPE_GUID:
&nbsp;&nbsp;&nbsp;&nbsp;if(n1->identifier.guid.data1 < n2->identifier.guid.data1 ||
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;n1->identifier.guid.data2 < n2->identifier.guid.data2 ||
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;n1->identifier.guid.data3 < n2->identifier.guid.data3 ||
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;strncmp((const char*)n1->identifier.guid.data4,
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;(const char*)n2->identifier.guid.data4, 8) < 0)
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return UA_ORDER_LESS;
&nbsp;&nbsp;&nbsp;&nbsp;if(n1->identifier.guid.data1 > n2->identifier.guid.data1 ||
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;n1->identifier.guid.data2 > n2->identifier.guid.data2 ||
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;n1->identifier.guid.data3 > n2->identifier.guid.data3 ||
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;strncmp((const char*)n1->identifier.guid.data4,
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;(const char*)n2->identifier.guid.data4, 8) > 0)
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return UA_ORDER_MORE;
&nbsp;&nbsp;&nbsp;&nbsp;break;

**fixed code:**
case UA_NODEIDTYPE_GUID: {
&nbsp;&nbsp;&nbsp;&nbsp;if(n1->identifier.guid.data1 < n2->identifier.guid.data1) {
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return UA_ORDER_LESS;
&nbsp;&nbsp;&nbsp;&nbsp;} else if(n1->identifier.guid.data1 > n2->identifier.guid.data1) {
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return UA_ORDER_MORE;
&nbsp;&nbsp;&nbsp;&nbsp;} else if(n1->identifier.guid.data2 < n2->identifier.guid.data2) {
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return UA_ORDER_LESS;
&nbsp;&nbsp;&nbsp;&nbsp;} else if(n1->identifier.guid.data2 > n2->identifier.guid.data2) {
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return UA_ORDER_MORE;
&nbsp;&nbsp;&nbsp;&nbsp;} else if(n1->identifier.guid.data3 < n2->identifier.guid.data3) {
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return UA_ORDER_LESS;
&nbsp;&nbsp;&nbsp;&nbsp;} else if(n1->identifier.guid.data3 > n2->identifier.guid.data3) {
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return UA_ORDER_MORE;
&nbsp;&nbsp;&nbsp;&nbsp;} else {
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;const int cmp = strncmp(
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;(const char*)n1->identifier.guid.data4,
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;(const char*)n2->identifier.guid.data4,
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;8
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;);
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;if(cmp < 0) return UA_ORDER_LESS;
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;if(cmp > 0) return UA_ORDER_MORE;
&nbsp;&nbsp;&nbsp;&nbsp;}
&nbsp;&nbsp;&nbsp;&nbsp;break;
}

Pro, 2019-07-19:
thanks for the issue report and the suggestion.

Could you please create a new Pull Request with your changes?

See https://hisham.hm/2016/01/01/how-to-make-a-pull-request-on-github-a-quick-tutorial/

Pro, 2019-07-24:
Fixed in #2904

PR #2904 (2019-07-24): Core: More intuitive node id ordering
Fix function UA_NodeId_order when compare UA_NodeId with identifierType equal UA_NODEIDTYPE_GUID

Fixes #2898
closed 2019-07-24 by Pro