open62541/open62541 #2899: Decoding error when adding multiple custom data types

m-fila, 2019-07-19:
## Description
Hi, I'm trying to add two custom data types basing on the [example](https://github.com/open62541/open62541/tree/master/examples/custom_datatype). The extra type is just structure with 4 float members. The problem is that I can't add more than 1 custom type at the same time --- reading the other results in `BadDecodingError`:
```bash
Point = 3.000000, 4.000000, 5.000000 
info/client	Error receiving the response with status code BadDecodingError
```
I tried adding both in 1 DataTypeArray:
**client.c**
```C
UA_DataType types[2];
types[0] = PointType;
types[0] = Point2Type;
UA_DataTypeArray customDataTypes = {NULL, 2, types};
cc->customDataTypes = &customDataTypes;
```
as well as in 2 arrays.:
**client.c**
```C
UA_DataType types[1];
types[0] = PointType;
UA_DataType types2[1];
types2[0] = Point2Type;
UA_DataTypeArray customDataTypes2 = {NULL, 1, types2};
UA_DataTypeArray customDataTypes = {&customDataTypes2, 1, types};
cc->customDataTypes = &customDataTypes;
```
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=REDUCED
```
Minimal example: 
[customtypes.zip](https://github.com/open62541/open62541/files/3411778/customtypes.zip)

* client.c  
* server.c         
* custom_datatype.h  
* custom_datatype2.h
* Makefile

## Checklist

Please provide the following information:

 - [ea6f481] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [  ] Self-contained code example attached
 - [ ] Critical issue

m-fila, 2019-07-20:
Closing. By mistake I provided both DataTypes with the same BinaryEncodingId

closed 2019-07-20 by m-fila

chuongvhn, 2019-12-23:
@m-fila : 

> UA_DataTypeArray customDataTypes = {NULL, 2, types};
> UA_DataTypeArray customDataTypes2 = {NULL, 1, types2};

Could you please help tell which is proper way to initialize value for second element of UA_DataTypeArray, typesSize ? Is is the size of custom datatype or something else ?

[https://github.com/open62541/open62541/blob/b074084510390601cda713ee9b2f3884411129ea/include/open62541/types.h#L1016](https://github.com/open62541/open62541/blob/b074084510390601cda713ee9b2f3884411129ea/include/open62541/types.h#L1016)

m-fila, 2019-12-23:
@chuongvhn , it's number of elements in  UA_DataType array. If you plan to write more than a few custom data types then I recommend using xml nodeset generator:
https://open62541.org/doc/current/nodeset_compiler.html
https://opcua.rocks/custom-information-models/