open62541/open62541 #2906: Install error with package manager

andri12321, 2019-07-23:
##  Description
Installing the packages from the daily build fails on (l)Ubuntu

## Background Information / Reproduction Steps
sudo add-apt-repository ppa:open62541-team/daily
sudo apt-get update
sudo apt-get install open62541-dev

## Additional Notes
The package open62541 installed without problems. Based on the error log the fault is probably that the shared libraries are in both packages.

## Error Log
client@spag-client2:~$ sudo apt-get install open62541-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libllvm4.0 linux-headers-4.10.0-28 linux-headers-4.10.0-28-generic
  linux-image-4.10.0-28-generic linux-image-extra-4.10.0-28-generic
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  open62541
The following NEW packages will be installed:
  open62541 open62541-dev
0 upgraded, 2 newly installed, 0 to remove and 4 not upgraded.
Need to get 0 B/3'236 kB of archives.
After this operation, 65.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package open62541:amd64.
(Reading database ... 212445 files and directories currently installed.)
Preparing to unpack .../open62541_1.0-0~201907221047~ubuntu16.04.1_amd64.deb ...
Unpacking open62541:amd64 (1.0-0~201907221047~ubuntu16.04.1) ...
Preparing to unpack .../open62541-dev_1.0-0~201907221047~ubuntu16.04.1_amd64.deb ...
Unpacking open62541-dev:amd64 (1.0-0~201907221047~ubuntu16.04.1) ...
dpkg: error processing archive /var/cache/apt/archives/open62541-dev_1.0-0~201907221047~ubuntu16.04.1_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libopen62541.so.1.0.0', which is also in package open62541:amd64 1.0-0~201907221047~ubuntu16.04.1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Processing triggers for libc-bin (2.23-0ubuntu11) ...
Errors were encountered while processing:
 /var/cache/apt/archives/open62541-dev_1.0-0~201907221047~ubuntu16.04.1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Pro, 2019-07-23:
Thanks for the report! Will have a look at it.

andri12321, 2019-07-31:
Thanks a lot! Works now

PR #2908 (2019-07-25): Debian: Do not copy lib into open62541-dev to avoid overriding open62541 package
Fixes #2906
commit 96a9cb19 (2019-07-25): Debian: Do not copy lib into open62541-dev to avoid overriding open62541 package

Fixes #2906
closed 2019-07-31 by andri12321