open62541/open62541 #2909: UA_Client makes repeated 824 requests after server shut down

lee-hemingway-adlink, 2019-07-23:
## Description

UA_Client, while acting as a subscriber, appears to intermittently get stuck in a loop making repeated 824 requests after the server is shut down. This loop prevents the state callback from triggering so our reconnection code does not run.

## Background Information / Reproduction Steps

1. Launch Prosys Simulation Server: https://downloads.prosysopc.com/opc-ua-simulation-server-downloads.php
2. Attach a stateCallback to the client which reconnects on UA_CLIENTSTATE_DISCONNECTED and will create a subscription and monitored item on UA_CLIENTSTATE_SESSION
4. The subscription should have an attached UA_Client_DataChangeNotificationCallback in which some action is performed on the data
3. Connect to the Prosys simulation server with all default client settings, this will trigger the UA_CLIENTSTATE_SESSION state above and create the subscription/monitored item
5. Call UA_Client_runAsync in a thread every 500ms while the client state is UA_CLIENTSTATE_SESSION
6. While the application is running and reading data, close the Prosys simulation server, it may be necessary to reopen and close Prosys repeatedly until the failure occurs

## Expected result

Most of the time, the state callback will trigger with the UA_CLIENTSTATE_DISCONNECTED state, which will then reconnect, trigger again with the UA_CLIENTSTATE_SESSION state and recreate the subscription/monitored item.

## Intermittent failure result

Sometimes intermittently, the client will get stuck in a loop of sending 824 requests. At this point UA_Client_runAsync is not being called as it is only called when the client is in state UA_CLIENTSTATE_SESSION. Therefore something internal to open62541 must be repeatedly causing 824 requests to be sent when the server is shut down. The state callback never triggers with UA_CLIENTSTATE_DISCONNECTED and so the client does not reconnect.

## Logs on intermittent failure

Log output has been truncated, the client will continue sending 824 requests until it is terminated.

```
[16:42:30] [UAClientWrapper.cpp:128>logOpen62541] [debug] [OPC UA Reader] [open62541.Client] Sending a request of type 824
[16:42:30] [UAClientWrapper.cpp:128>logOpen62541] [debug] [OPC UA Reader] [open62541.Client] Sending a request of type 629
[16:42:30] [UAClientWrapper.cpp:138>logOpen62541] [warning] [OPC UA Reader] [open62541.Client] Received Publish Response with code BadNoSubscription
[16:42:30] [UAClientWrapper.cpp:128>logOpen62541] [debug] [OPC UA Reader] [open62541.Client] Sending a request of type 824
[16:42:30] [UAClientWrapper.cpp:138>logOpen62541] [warning] [OPC UA Reader] [open62541.Client] Received Publish Response with code BadTimeout
[16:42:30] [UAClientWrapper.cpp:128>logOpen62541] [debug] [OPC UA Reader] [open62541.Client] Sending a request of type 824
[16:42:30] [UAClientWrapper.cpp:138>logOpen62541] [warning] [OPC UA Reader] [open62541.Client] Received Publish Response with code BadTimeout
[16:42:30] [UAClientWrapper.cpp:128>logOpen62541] [debug] [OPC UA Reader] [open62541.Client] Sending a request of type 824
[16:42:30] [UAClientWrapper.cpp:138>logOpen62541] [warning] [OPC UA Reader] [open62541.Client] Received Publish Response with code BadTimeout
[16:42:30] [UAClientWrapper.cpp:128>logOpen62541] [debug] [OPC UA Reader] [open62541.Client] Sending a request of type 824
[16:42:30] [UAClientWrapper.cpp:138>logOpen62541] [warning] [OPC UA Reader] [open62541.Client] Received Publish Response with code BadTimeout
```

Used CMake options:

-DUA_NO_AMALGAMATION

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v0.3
 - [ ] Other OPC UA SDKs used (client or server): Prosys Simulation Server
 - [ ] Operating system: Debian Stretch
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-07-23:
I just checked the ids:

* 824 = PublishRequest
* 629 = ReadRequest

The output you see comes from here:
https://github.com/open62541/open62541/blob/d0c2e5989f56dab8bf24b8769e9ef3b8e9aa75c4/src/client/ua_client_subscriptions.c#L667

And since the backgroundConnectivityCheck comes after the publish, the client never detects the timeout correspondingly:

https://github.com/open62541/open62541/blob/d0c2e5989f56dab8bf24b8769e9ef3b8e9aa75c4/src/client/ua_client.c#L528

So I guess there is an additional check missing.

```
    if(response->responseHeader.serviceResult == UA_STATUSCODE_BADTIMEOUT) {
        if (client->config.inactivityCallback)
            client->config.inactivityCallback(client);
        return;
    }
```

Should come in here:
https://github.com/open62541/open62541/blob/d0c2e5989f56dab8bf24b8769e9ef3b8e9aa75c4/src/client/ua_client_subscriptions.c#L665

Let me try to create a PR

Pro, 2019-07-23:
@lee-hemingway-adlink can you check my PR #2910 if that fixes your issue?

Also note that we will soon release v1.0 (which probably also has that issue, will create another PR)

lee-hemingway-adlink, 2019-07-23:
Thank you very much for looking into this. It now appears to be getting to the point where the stateCallback triggers with UA_CLIENTSTATE_SESSION, the subscription and monitored item get created and then I get a segfault with the following output:
```
[13:24:51] [OPCUAReader.cpp:331>registerOPCUAMonitoredItem] [debug] [OPC UA Reader] Successfully created an OPC UA monitored item - SubscriptionID: [2] NodeId: [5:Random1]
*** Error in `/aea-opc-ua-client': free(): invalid pointer: 0x00007f6b38000128 ***
[13:24:51] [UAClientWrapper.cpp:134>logOpen62541] [warning] [OPC UA Reader] [open62541.Client] Received Timeout for Publish Response
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bfb)[0x7f6c2a486bfb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76fc6)[0x7f6c2a48cfc6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7780e)[0x7f6c2a48d80e]
/aea-opc-ua-client(+0x9366c)[0x5561c390e66c]
```
Unfortunately this does not seem to be ocurring in GDB or Valgrind so I was unable to get symbol names.

Pro, 2019-07-24:
Can you try to use clang with adress sanitizer?

Easiest setup:
- Use clang to build
- Add these CMake settings somewhere in your CMakeLists.txt

```
    if(BUILD_TYPE_LOWER_CASE STREQUAL "debug")
        if("x${CMAKE_C_COMPILER_ID}" STREQUAL "xClang" AND NOT UA_ENABLE_UNIT_TESTS_MEMCHECK)
            # Add default sanitizer settings when using clang and Debug build.
            # This allows e.g. CLion to find memory locations for SegFaults
            message(STATUS "Sanitizer enabled")
            set(SANITIZER_FLAGS "-g -fno-omit-frame-pointer -gline-tables-only -fsanitize=address -fsanitize-address-use-after-scope -fsanitize-coverage=trace-pc-guard,trace-cmp -fsanitize=leak -fsanitize=undefined")
            set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${SANITIZER_FLAGS}")
            set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${SANITIZER_FLAGS}")
        endif()
    endif()
```

If that doesn't give use a corresponding symbol name, you need to create a minimal verifiable example which we can use to reproduce the issue.

lee-hemingway-adlink, 2019-07-24:
The segfault appears to have been an issue with our own code which I have now fixed.

So I can confirm that your change fixes the repeated 824 request issue, thank you very much for your help :)

PR #2910 (2019-07-24): Client: Add timeout check for publish response
Fixes #2909
commit f93e695f (2019-07-24): Client: Add timeout check for publish response

Fixes #2909
PR #2911 (2019-07-24): Client: Add timeout check for publish response
Fixes #2909

Port of #2910 to 1.0 branch
commit 1994c118 (2019-07-24): Client: Add timeout check for publish response

Fixes #2909
closed 2019-07-24 by lee-hemingway-adlink