open62541/open62541 #2913: Get all changed monitored items of subscription at the same time

je1h, 2019-07-23:
As far as I can see from the OPC-UA specifications, on every publishing interval OPC-UA server sends one NotificationMessage containing all Notifications of all changes (I'm monitoring variable values) it has sampled in queue:

> a) Subscriptions have a set of MonitoredItems assigned to them by the Client. MonitoredItems generate Notifications that are to be reported to the Client by the Subscription (see 5.12.1 for a description of MonitoredItems).
> b) Subscriptions have a publishing interval. The publishing interval of a Subscription defines the cyclic rate at which the Subscription executes. Each time it executes, it attempts to send a NotificationMessage to the Client. NotificationMessages contain Notifications that have not yet been reported to Client.

But as I found in open62541 documentation, its subscription methods `UA_Client_MonitoredItems_createDataChange` and `UA_Client_MonitoredItems_createDataChanges` work on "callback per every single monitored item" basis. Item by item.

Is there a way to get all the monitored items changes of one publishing interval in a bulk?

- 1-st publishing interval: changed values of items 1, 2, 3

- 2-nd publishing interval: changed values of items 2, 4, 5

- etc...

schroeder-, 2019-07-23:
I think it is not possible. But you can use the same callback for all MonitoredItems and add a UA_Client_addTimedCallback with UA_DateTime_nowMonotonic as time Parameter. So you will get the Callback after the current NotficationMessage. This only works if you use UA_Client_run_iterate with timeout 0.

je1h, 2019-07-26:
Thanks for your answer!
I do `UA_Client_run_iterate(client, 0);` after subscriptions have been set up.
But the `UA_Client_addTimedCallback` timer I set up in callback never fires.

teccheck, 2025-02-25:
The answer by @schroeder- describes the intended way to have one callback. The timer issue should be fixed since there is a new event loop now

closed 2025-02-25 by teccheck