open62541/open62541 #2918: Can't create a server ExtensionObject Variable Node anymore with master

xallaume, 2019-07-24:
## Description 
Former code creating a server ExtensionObject Variable Node working with release 0.3 doesn't work anymore with master. Basically, it seems that extension.Object.encoding is implied. Now doing:

```cpp
const char * strStringTest = "String Payload as a ByteString extension";
    UA_ExtensionObject myExtensionObject;
    UA_ExtensionObject_init( &myExtensionObject );
    UA_NodeId nodeId;
    UA_NodeId_init( &nodeId );
    nodeId = UA_NODEID_STRING(5, generateCharArrayFromString("StringTestNode"));
    myExtensionObject.content.encoded.typeId = nodeId;
    myExtensionObject.encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
    myExtensionObject.content.encoded.body.length = strlen( strStringTest );
    myExtensionObject.content.encoded.body.data = ( UA_Byte * ) strStringTest;

UA_Variant_setScalar( &attr.value, &myExtensionObject, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT] );
        attr.dataType = UA_TYPES[UA_TYPES_EXTENSIONOBJECT].typeId;

...

UA_StatusCode ret = UA_Server_addVariableNode(server, myVariableNodeId, parentNodeId,
                        parentReferenceNodeId, myTestName,
                        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);

This last function returns 2151481344 (BadNotSupported) due to myExtensionObject.encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING in static UA_StatusCode copyCommonVariableAttributes(UA_VariableNode *node, const UA_VariableAttributes *attr).
```

Any hint on how to solve this?

## Background Information / Reproduction Steps

Used CMake options:
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=REDUCED

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

xallaume, 2019-07-25:
I'm probably not clear enough. Is this the case?

xallaume, 2019-08-01:
I spent a bit more time on it, I can see the problem also happens in 1.0-RC1.

I can see that the test for UA_ExtensionObject_encodeDecodeShallWorkOnExtensionObject has been commented, and I can't seem to find an example for writing a Byte String in a UA_Extension_Object in the example folder either.

xallaume, 2019-08-01:
I spent a bit more time on it, I can see the problem also happens in 1.0-RC1.

The following code exposes the problem and is self sustained:

```cpp
#include <iostream>
#include <vector>
#include <string>
using namespace std;
#include "open62541.h"

static void addVariableTestNode( UA_Server * server, string nodeName, string type )
{
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    
    const char * strStringTest = "String Payload as a ByteString extension";
    UA_ExtensionObject myExtensionObject;
    UA_ExtensionObject_init( &myExtensionObject );
    myExtensionObject.encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
    myExtensionObject.content.encoded.typeId = UA_NODEID_STRING( 5, &string( "StringTestNode" )[0u] );
    myExtensionObject.content.encoded.body.length = strlen( strStringTest );
    myExtensionObject.content.encoded.body.data = ( UA_Byte * ) strStringTest;
    
    attr.dataType = UA_TYPES[UA_TYPES_EXTENSIONOBJECT].typeId;        
        UA_Variant_setScalar( &attr.value, &myExtensionObject, &UA_TYPES[UA_TYPES_EXTENSIONOBJECT] );
    
    std::vector<char> writable(nodeName.begin(), nodeName.end());
    writable.push_back('\0');
    char * nodeNam = &writable[0];

    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    attr.description = UA_LOCALIZEDTEXT(&string("en-US")[0u],nodeNam);
    attr.displayName = UA_LOCALIZEDTEXT(&string("en-US")[0u],nodeNam);

    /* Add the variable node to the information model */
    UA_NodeId myVariableNodeId = UA_NODEID_STRING( 1, nodeNam );
    UA_QualifiedName myTestName = UA_QUALIFIEDNAME( 1, nodeNam );
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT);
    UA_StatusCode ret = UA_Server_addVariableNode(server, myVariableNodeId, parentNodeId,
                        parentReferenceNodeId, myTestName,
                        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);

    if (ret != UA_STATUSCODE_GOOD)
        cout << "Houston, we have a problem... with " << nodeName << endl;
}

UA_Boolean running = true;

int main(void) {
    UA_Server * server = UA_Server_new();
    UA_ServerConfig_setDefault( UA_Server_getConfig( server ) );       

    addVariableTestNode( server, "ExtensionObjectTestNode", "ExtensionObject" );
        
    UA_StatusCode retval = UA_Server_run(server, &running);
    
    UA_Server_delete(server);
    return (int)retval;
}
```

jpfr, 2019-08-05:
This is fixed in #2964.

xallaume, 2019-08-06:
Works as expected now. Thank you very much!!!

closed 2019-08-16 by jpfr