open62541/open62541 #2919: Writing of binary encoded euromap structures leads to UA_STATUSCODE_BADENCODINGLIMITSEXCEEDED

hse04004, 2019-07-24:
## Description
I am trying to build an euromap client for communicating with an OpcUa server that supports euromap 77 and 83 types.
When I try to read structures, binary decoding of euromap structures work perfectly fine.
E.g. decoding StandstillReasonType:
`UA_Variant lVariant;`
`UA_StatusCode lStatus = UA_Client_readValueAttribute(spClient, iNodeId, &lVariant);`
`UA_ExtensionObject *lpVariantData = reinterpret_cast<UA_ExtensionObject*>(lVariant.data);`
`UA_StandstillReasonType lStandstillReason;`
`lStatus = UA_StandstillReasonType_decodeBinary(&lpVariantData->content.encoded.body, &offset, &lStandstillReason);`

If I try to encode and write a similar structure, encoding works but when I try to write I get a BADENCODINGLIMITSEXCEEDED.
E.g. encoding StandstillReasonType:
`UA_Variant lVariant;`
`size_t offset = 0;`
`UA_StandstillReasonType lStandstillReason;`
`lStandstillReason.id = UA_String_fromChars("test");`
`UA_StatusCode lStatus = UA_Client_readValueAttribute(spClient, iNodeId, &lVariant);`
`UA_ExtensionObject *lpVariantData = reinterpret_cast<UA_ExtensionObject*>(lVariant.data);`
`lStatus = UA_StandstillReasonType_encodeBinary(&lStandstillReason, &lpVariantData->content.encoded.body.data, &lpVariantData->content.encoded.body.data[lpVariantData->content.encoded.body.length]);` All fine to this point
`lStatus = UA_Client_writeValueAttribute(spClient, lNodeId, &lVariant);` lStatus == UA_STATUSCODE_BADENCODINGLIMITSEXCEEDED here.

I could not find an example of how to proper use encodeBinary of predefined euromap structures.
I am not sure if i am missing something here, but shouldn't that work?

Thanks for your support.

## Background Information / Reproduction Steps

git clone https://github.com/Pro/open62541.git
git checkout test/euromap
git submodule update --init --recursive
Used CMake options:

```bash
cmake .. -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=YES -DUA_ENABLE_SUBSCRIPTIONS=YES -DUA_ENABLE_METHODCALLS=YES -DUA_ENABLE_PUBSUB=YES -DUA_ENABLE_PUBSUB_DELTAFRAMES=YES -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=YES -DUA_ENABLE_HISTORIZING=YES -DUA_BUILD_EXAMPLES=YES -DUA_ENABLE_AMALGAMATION=NO -G "Visual Studio 15 2017"
```
Build with Visual Studio

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): test/euromap
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: win
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-07-24:
Using the encoding and decoding methods is not recommended.

Are you checking the return value of `UA_StandstillReasonType_encodeBinary`?

The easier way would be to let the stack handle it. E.g. what you can try is the following:

```
UA_StandstillReasonType lStandstillReason;
UA_StandstillReasonType_init(&lStandstillReason);
lStandstillReason.id = UA_String_fromChars("test");

UA_Variant newVar;
UA_Variant_init(&newVar);

// make sure you are using the correct types array. These values are just exemplary
UA_Variant_setScalar(&newVar, &lStandstillReason, &UA_TYPES_EUROMAP88[UA_TYPES_EUROMAP88_STANDSTILLREASONTYPE]);

lStatus = UA_Client_writeValueAttribute(spClient, lNodeId, &newVar);
```

Does that work for you?

hse04004, 2019-07-25:
Thank you for the quick response.

The `UA_StandstillReasonType_encodeBinary` returned good. The error occured not until `UA_Client_writeValueAttribute`.

Anyways `UA_Variant_setScalar` works fine.

Thank you.

closed 2019-07-25 by Pro