open62541/open62541 #2920: Generated custom data-types inside a method call

starki93, 2019-07-24:
## Description
For my project I used the following tutorial to implement custom data-types.

https://opcua.rocks/custom-information-models/

It’s possible to read all custom structure description with the external tool UAExpert and use this tool to set variables und call methods with my custom data-type. But I have some internal problems to use the custom data-types inside a method-call. The input is received as an ExtensionObject and I am not able to decode it.

I already tried the reduced and full namespace option. Build the project with UA_ENABLE_AMALGAMATION option and without.

The custom structure has one UA_String and three UA_Int32.
```
   <DataType` SymbolicName="YR:ToolState" BaseType="ua:Structure">
        <Description>Sound of an animal</Description>
        <Fields>
            <Field Name="Name" DataType="ua:String">
                <Description>Sound Verb</Description>
            </Field>
            <Field Name="Dout" DataType="ua:Int32">
                <Description>URL to an audio file</Description>
            </Field>
            <Field Name="Din" DataType="ua:Int32">
                <Description>URL to an audio file</Description>
            </Field>
            <Field Name="Aout" DataType="ua:Int32">
                <Description>URL to an audio file</Description>
            </Field>
        </Fields>
    </DataType>
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.0
 - [ ] Other OPC UA SDKs used (client or server): UAExpert 
 - [ ] Operating system: linux

Pro, 2019-07-24:
How did you try to access your custom datatype?

If you followed the tutorial (which was written by me), you will end up with a NodeSet2.xml and a .bsd and .csv file.

You need these files as input for the nodeset compiler macro. An example is given here:

https://github.com/open62541/open62541/blob/b0971ef585f0512521ef54a71e9ee0efc40d8ca6/examples/nodeset/CMakeLists.txt#L46

You normally need the FULL NS0 otherwise some parent nodes will be missing.

The macro will generate the necessary structs which you can directly use in the code, i.e., `UA_ToolState`.

E.g. see this code example here:
https://github.com/open62541/open62541/blob/b0971ef585f0512521ef54a71e9ee0efc40d8ca6/examples/nodeset/server_testnodeset.c#L47

Does that answer your question?

starki93, 2019-07-25:
At CMake file i used the following code:
```cmake
option(UA_ENABLE_PUBSUB "" ON)
set(UA_NAMESPACE_ZERO FULL CACHE STRING "" FORCE)

set(open62541_TOOLS_DIR "${CMAKE_CURRENT_SOURCE_DIR}/open62541/tools")
set(open62541_NODESET_DIR "${CMAKE_CURRENT_SOURCE_DIR}/open62541/deps/ua-nodeset")

add_subdirectory(open62541)

ua_generate_nodeset_and_datatypes(
        NAME "yr"
        FILE_CSV "${CMAKE_CURRENT_SOURCE_DIR}/model/yrModel.csv"
        FILE_BSD "${CMAKE_CURRENT_SOURCE_DIR}/model/yr.Types.bsd"
        NAMESPACE_IDX 2
        FILE_NS "${CMAKE_CURRENT_SOURCE_DIR}/model/yr.NodeSet2.xml"
        OUTPUT_DIR "${CMAKE_CURRENT_BINARY_DIR}/open62541/src_generated/open62541"
        INTERNAL
)
```

The method is added like this:

```cpp
static void
addMethod(UA_Server *server) {
    UA_Argument inputArgument;
    UA_Argument_init(&inputArgument);

    inputArgument.description = UA_LOCALIZEDTEXT((char*)"en-US", (char*)"set tool");
    inputArgument.name = UA_STRING((char*)"input");
    inputArgument.dataType = UA_TYPES_YR[UA_TYPES_YR_TOOLSTATE].typeId;
    inputArgument.valueRank = UA_VALUERANK_ONE_DIMENSION;

    UA_MethodAttributes helloAttr = UA_MethodAttributes_default;
    helloAttr.description = UA_LOCALIZEDTEXT((char*)"en-US",(char*)"set tool");
    helloAttr.displayName = UA_LOCALIZEDTEXT((char*)"en-US",(char*)"set tool");
    helloAttr.executable = true;
    helloAttr.userExecutable = true;

    UA_Server_addMethodNode(server, UA_NODEID_NUMERIC(1,1000),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASORDEREDCOMPONENT),
                            UA_QUALIFIEDNAME(1, (char*)"test method"),
                            helloAttr, &methodCallback,
                            1, &inputArgument, 0, nullptr, nullptr, nullptr);
}
```

And the call back is:

```cpp
static UA_StatusCode
methodCallback(UA_Server *server,
                         const UA_NodeId *sessionId, void *sessionHandle,
                         const UA_NodeId *methodId, void *methodContext,
                         const UA_NodeId *objectId, void *objectContext,
                         size_t inputSize, const UA_Variant *input,
                         size_t outputSize, UA_Variant *output) {

    printf("received command with input size:%lu \n", inputSize);

    for(size_t i = 0; i < inputSize; i++) {
        printf("typename: %s\n", input[i].type->typeName);
    }

    return UA_STATUSCODE_GOOD;
}
```
I get the following output if i call this method with an arraysize 3:

```
[2019-07-25 09:18:35.762 (UTC+0200)] trace/network	Connection 5 | Activity on the socket
[2019-07-25 09:18:35.762 (UTC+0200)] trace/network	Connection 5 | Received a packet.
[2019-07-25 09:18:35.762 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Decrypting chunk
[2019-07-25 09:18:35.762 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Chunk size before and after decryption: 168, 168
[2019-07-25 09:18:35.762 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Decrypted and verified chunk with request id 57 and sequence number 107
[2019-07-25 09:18:35.762 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Sequence Number processed: 107
[2019-07-25 09:18:35.762 (UTC+0200)] trace/channel	Connection 5 | SecureChannel 1 | Process a MSG
[2019-07-25 09:18:35.762 (UTC+0200)] debug/session	Connection 5 | SecureChannel 1 | Session ns=1;g=cfd85efa-5d9d-cd24-bddb-5dd518919959 | Processing CallRequest
received command with input size:1 
typename: ExtensionObject
```

starki93, 2019-07-25:
Now i set value-rank to scalar and modified the callback to:
```cpp
    UA_ToolState *p = (UA_ToolState *)input[0].data;
    printf("din:%d dout:%d aout:%d\n", p->din, p->dout, p->aout);
```

The output i get is:
```
din:0 dout:15027 aout:0
```
But all inputs are set to "1".

teccheck, 2025-02-25:
This should be solved now. Feel free to reopen if needed

closed 2025-02-25 by teccheck