open62541/open62541 #2924: Compiled nodesets do not initialize the variable type, UA Expert can not write to variable

Pro, 2019-07-25:
UaExpert does not allow to write to variables if the current value is NULL. Also the variable data type is not displayed.
Like this UaExpert can not be used to write to these variables and you need to use another client for that.

If I remember correctly, this changed here:
https://github.com/open62541/open62541/commit/71d206c202a8612d885e7f78821f177238a4fd81#diff-20d3724d2a13ae0740cbdb65ea6d0ea8L254

It removed the dummy value initialization. Before that every node was set to an empty value different than null with the effect that the data type was at least initialized.

I remember that I had to remove this, otherwise some nodesets could not be initialized. But we need to fix this another way.

We need to somehow fix this NULL initialization so that it is possible to write to such nodes using UaExpert.

ccvca, 2019-07-25:
From my point of view it does not make sense to initialize all values "neutral" just to support UaExpert.
To solve this problem I would suggest two possible solutions:
* Introduction of a function to fill a node/subtree along hierarchical references with neutral values.
* Unified Automation bug report that UaExpert cannot write to null values with the datatype specified in the DateType attribute. This addresses also their (C++)-SDK especially when dealing with imported namespaces from XML-files.

Pro, 2019-08-28:
I posted a question on the UaExpert forum:
https://forum.unified-automation.com/post11520.html

Response by the support team:

> this is expected behavior. Typically when a node is being subscribed it already has a value (and a value always has a type). The DAView is obtaining the values datatype form the received value (received during first update). However if your node has no value the UaExpert will assume its type is "NULL", because it never has received a value. Reading the attribute (as suggested by you) makes no sense, because the node can change the valued datatype during runtime, so besides creating unnessecary load (by manually reading data type attribute), the datatype may still be different when being used. And of course you can write the value, just need to use the attribute window for writing.
> 
> In any case you should take a close look at the Server, why is it having nodes that have no value? That makes not much sense.

Summarized, every variable should be initialized with a specific value. The question now remains:
@jpfr Should we automatically create a empty not-null dummy value in the nodeset compiler, or should we leave this up to the stack user?

jpfr, 2021-11-23:
We now auto-generate a value when the node is added and has only a NULL value.
If this is not allowed.

closed 2021-11-23 by jpfr