open62541/open62541 #2929: Execution time UA_Server_run_iterate() higher each second

SebastianGraeber, 2019-07-26:
## Description

In my application (100 Hz with UA_Server_run_iterate) I observed some unusual jitter. Normal execution time is around 2.8 ms, but each 100 samples the execution time is around 7.0 ms. I benchmarked my application to see the execution time of each block and the method **UA_Server_run_iterate(server, false)** causes this higher latency each 100 samples.

Beside the latency issue the application runs very well.

What happens inside this method that might cause this latency? How can this behaviour be fixed?

Thank you.

Regards,
Sebastian

## Background Information / Reproduction Steps

OS is CentOS7 with 5.2-RT kernel (compiled by myself)
The application runs as root with real-time privileges, pinned to a dedicated CPU core (isolated in kernel boot command line) and locked memory (mlockall()).
I use the amalgamated version (*.c/*h) without multithreading and compiled with -m32.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0
 - [x] Other OPC UA SDKs used (client or server): -
 - [x] Operating system: CentOS 7 with 5.2-rt kernel / CPU isolated and app pinned to that CPU
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-07-27:
We have some internal cleanups (test for session timeouts, etc.) that are performed periodically.
Are you the increased duration is exactly every 100 iterations?

7ms seems excessively long for the cleanup.
Maybe this is related to the networking layer.

Your best chance is to add some custom instrumentation to see where the delay comes from on your system.

SebastianGraeber, 2019-07-30:
I re-created the set-up on an other machine and the latency is gone. I guess it's the network interface. I will have a deeper look into this.

Issue closed.

SebastianGraeber, 2019-08-05:
I had a closer look into this issue and I figured out that the latency issue only occurs with two active network interface cards.
I walked down to the method "processChunk(...)" and it seems that its there or deeper inside the code.

Maybe you guys have an idea how I might bring down the latency with two active network interface cards. I need both of them as I want to seperate both networks (internal and externel network for customers).

Thank you.

Regards,
Sebastian

teccheck, 2025-02-25:
With the new event loop landed, this should be fixed

closed 2025-02-25 by teccheck