open62541/open62541 #2930: Unresolved external symbol errors

JustinKieu, 2019-07-26:
## Description
I'm running into issues with trying to build a simple server with the AMALGAMATION flag enabled. It gives me these linking errors despite linking the open62541.lib

```
1>LINK : warning LNK4098: defaultlib 'LIBCMTD' conflicts with use of other libs; use /NODEFAULTLIB:library
1>myServer.obj : error LNK2019: unresolved external symbol UA_ServerConfig_new_default referenced in function main
1>myServer.obj : error LNK2019: unresolved external symbol UA_ServerConfig_delete referenced in function main
1>open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_DateTime_now referenced in function UA_random_seed
1>open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_DateTime_localTimeUtcOffset referenced in function UA_CertificateVerification_AcceptAll
1>open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_DateTime_nowMonotonic referenced in function UA_Server_run_iterate
1>open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_ServerNetworkLayerTCP referenced in function UA_ServerConfig_addNetworkLayerTCP
1>open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_ClientConnectionTCP referenced in function UA_ClientConfig_setDefault
1>open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_ClientConnectionTCP_poll referenced in function UA_ClientConnectionTCP_poll_callback
1>open62541.lib(open62541.obj) : error LNK2019: unresolved external symbol UA_ClientConnectionTCP_init referenced in function UA_ClientConfig_setDefault
1>C:\Users\SESA538878\Downloads\open62541SINGLE2\open62541-master\build\x64\Debug\OPCUASINGLE.exe : fatal error LNK1120: 9 unresolved externals
```
I'm not too sure where I'm going wrong.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<AMALGAMATION = ON> 
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Operating system: Windows

JustinKieu, 2019-07-26:
I tested by downloading the 0.3 single distribution code and was able to run the tutorial server. I'm not sure if there's an error when using the amalgamation flag.

Pro, 2019-07-27:
How exactly did you build the library and how does you command line look to build the tutorial?
Please give use the full list of commands, especially CMake settings and make flags.

jpfr, 2019-07-27:
We are in the nasty habit of rewriting the /MT /MD compiler flags in cmake.
This is fixed as part of #2874 .

With this, we can link against library dependencies from vcpkg.

JustinKieu, 2019-07-29:
Other than turning on the amalgamation flag, I left every other flag on as default. I used the cmake GUI. I did not turn on the build_shared_libs flag. 
To build the library:
1. I opened the solution on visual studio that was generated from CMake.
2. I built the solution.
3. checked in the build folder for single source generated c and h files for open62541.
4. I created a new project and added in both the h and c files.
5. I created the myServer.c file.
6. I linked the library to the new project.
7. I then just tried to debug on visual studio. Thats when I ran into the errors from above.

JustinKieu, 2019-07-29:
I also downloaded the single file relase tag: 9095278e9f (the most recent one) and am coming into this error.
[image]
I'm not sure if its something on my end. I just created a  new project and added both the c and h files to the project. I then just used the simple server tutorial again to try and build.

jpfr, 2019-07-29:
Hmm. That's not good.
Which version of Visual Studio is that?

JustinKieu, 2019-07-29:
It is Microsoft Visual Studio Professional 2019 Version 16.1.5

JustinKieu, 2019-08-19:
@jpfr  Was this fixed with #2874 ? I tested again recently and I'm still coming up with the same unresolved symbols

rdvojmoc, 2019-08-29:
I'm experiencing same issue in VS 2017 v15.9.7. Part of my CMakeLists.txt which is responsible for compiling open62541 library:
```
# -----------------------------------
# open62541 specific settings - BEGIN
# -----------------------------------
set(UA_ENABLE_AMALGAMATION ON CACHE BOOL "" FORCE)
set(UA_LOGLEVEL 300) #INFO
add_subdirectory ("open62541")

set_source_files_properties("${PROJECT_BINARY_DIR}/open62541/open62541.c" PROPERTIES GENERATED TRUE)
set(${PROJECT_NAME}_SRCS ${${PROJECT_NAME}_SRCS} "${PROJECT_BINARY_DIR}/open62541/open62541.c")
include_directories("${PROJECT_BINARY_DIR}/open62541/")
# -----------------------------------
# open62541 specific settings - END
# -----------------------------------

add_dependencies(${PROJECT_NAME} open62541 open62541-amalgamation-source open62541-amalgamation-header) 

target_link_libraries(${PROJECT_NAME} open62541)
```
Compile time error:
'UA_ServerConfig_delete': identifier not found 	

What could be done to resolve this.

Lemonbonbon, 2023-10-20:
I have the same problem in VS 2022. Is there a solution?

NoelGraf, 2025-02-18:
You need to link your project against the Winsock2 (ws2_32) and IP Helper (Iphlpapi) libraries.
In a CMake project, add the following line:

`target_link_libraries(<target> ws2_32 Iphlpapi)`

closed 2026-10-20 by o6-nbeier