open62541/open62541 #2932: Is there a macro for version dependent compilation

tobiaslocker, 2019-07-28:
## Description

I have created a [C++ wrapper](https://github.com/tobiaslocker/open62541pp) for open62541, currently based on `v0.3.0`. I want to implement support for other versions, looking forward for `v1.0.0`, and avoid passing a macro to the build if you guys already have done so.

Pro, 2019-07-28:
Thanks for your work. Do you already know about #2067, especially https://github.com/bazza2jw/open62541Cpp

(ping @bazza2jw)

IMHO it would make sense to join your efforts in the C++ wrapper development.

> and avoid passing a macro to the build

What exactly do you mean by that?

tobiaslocker, 2019-07-28:
Yes I am aware of the issue and the work of bazza2jw. I decided to start a different wrapper for my project based on some design decisions (e.g. using raw pointers and open62541 types in c++ api). But any contribution is welcome and highly appreciated!

Regarding my issue, given an example:
I wonder if you have a version macro, so i could change implementation to something like

```
#if OPEN_62541_VERSION >= 0x010000 // Qt handles version dependent compilation like this
  // v1.0.0
#else 
  // v0.3.0 implementation
#endif
  
```

Pro, 2019-07-28:
It depends how you integrate our library.

If you are using our project as a submodule in CMake, then you can use these variables:
https://github.com/open62541/open62541/blob/347a486f538ffe7c076e6e6e1dd0d2cd578efeff/CMakeLists.txt#L35

Starting with version 1.0 we also improved the CMake FindPackage stuff where you can indicate the version, and also get the version which was found by CMake:
https://github.com/open62541/open62541/blob/6aa4d8d9d2f76b0ba95acba06345ae38c3e4c068/doc/installing.rst#manual-installation

Ultimatively you can always use the define in our config.h inside the source code:

https://github.com/open62541/open62541/blob/347a486f538ffe7c076e6e6e1dd0d2cd578efeff/include/open62541/config.h.in#L11

That's probably what you are looking for. Sorry, I was a bit confused, since the correct term is define and not macro.

Please re-open if the question is not solved.

closed 2019-07-28 by Pro

tobiaslocker, 2019-07-28:
The correct term is [Object-like Macro](https://gcc.gnu.org/onlinedocs/cpp/Object-like-Macros.html#Object-like-Macros) ;)

Yes, this was what i was looking for, thanks! Keep up the work!