open62541/open62541 #2938: Python version support

mlgiraud, 2019-07-30:
Official Python 2 support will end in 2020.
Should we continue to support Python 2, or make the change to only support Python 3 from now on?
On most machines python3 should be already installed by default, and installing a desired version is very easy by using tools such as pyenv.

A further question arising would be which python3 version to support.
The current newest version is 3.7.4, with 3.8 and 3.9 being in development. 3.8 is scheduled for release in October this year.
Ubuntu 16 LTS only has python up to 3.5 in the package manager, while Ubuntu 18 LTS has python up to 3.6,

Although imho i would not find this a limiting factor, since managing python versions with tools like pyenv and pipenv is really easy.

What are your thoughts on this? @jpfr @Pro

jpfr, 2019-07-30:
Imho it takes only minimal effort to support both.
Since we have both py2 and py3 in Travis.
Let's revisit when py2 maintenance has ended and/or it becomes more effort to support both.

Pro, 2019-07-30:
Same here. Since we already support both, we should keep that for now, until py2 is EOL.

Same for py3, we should keep the lowest possible and highest necessary version support. As long as we do not really need a feature from 3.8, let's keep support for older.
Seems travis is currently using 3.6 by default.

Pro, 2019-08-30:
I'll close this for now. We can reopen the discussion in end of 2020

closed 2019-08-30 by Pro