open62541/open62541 #2939: Critical stuck execution problem when creating monitored items

cookiejarlid, 2019-07-30:
## Description

Calls to ```UA_Client_MonitoredItems_createDataChanges``` will get stuck and not call the user supplied callback or return to the application code under certain conditions, more details below.

## Background Information / Reproduction Steps

I've noticed this issue when using the publicly available opc-ua server ```opc.tcp://milo.digitalpetri.com:62541/milo```

To reproduce the problem, simply create a monitored item using ```UA_Client_MonitoredItems_createDataChanges``` for a node id that doesn't exist on that server, e.g.: ```ns=2;s=Something```

Here are the last log entries (```UA_LOGLEVEL``` set to 100) before execution gets stuck:

```
[2019-07-30 11:21:58.035 (UTC+0100)] trace/client	Client connect iterate
[2019-07-30 11:21:58.036 (UTC+0100)] debug/client	Sending a request of type 749
[2019-07-30 11:21:58.534 (UTC+0100)] trace/channel	Connection 11 | SecureChannel 34330 | Decrypting chunk
[2019-07-30 11:21:58.534 (UTC+0100)] trace/channel	Connection 11 | SecureChannel 34330 | Chunk size before and after decryption: 83, 83
[2019-07-30 11:21:58.534 (UTC+0100)] trace/channel	Connection 11 | SecureChannel 34330 | Decrypted and verified chunk with request id 16 and sequence number 6
[2019-07-30 11:21:58.535 (UTC+0100)] trace/channel	Connection 11 | SecureChannel 34330 | Sequence Number processed: 6
[2019-07-30 11:21:58.535 (UTC+0100)] debug/client	Decode a message of type CreateMonitoredItemsResponse
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Other OPC UA SDKs used (client or server): opc.tcp://milo.digitalpetri.com:62541/milo
 - [x] Operating system: Linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2019-07-30:
Hello @cookiejarlid 

can you reproduce the behaviour in a debugger?
We need to see a backtrace to help you.

Either the `bt` output in gdb or a screenshot of the Visual Studio debugger should be enough.

cookiejarlid, 2019-08-01:
I haven't been able to reproduce the problem since the day I opened the issue on.
My guess is that this particular server was replying something that caused open62541 to become stuck (possibly inside a loop?) while decoding a ```CreateMonitoredItemsResponse``` message.
I'll keep an eye on it and I'll post more info whenever it happens again.

Zvicii, 2020-04-01:
I have met the same problem.
I build a client and a server using open62541.
but when i try to create monitoreditems using UA_Client_MonitoredItems_createDataChanges, if the number is too large(5000 string nodeid), it gets stuck and doesnt return.the server would also stuck in its loop and cant be sopped by setting 'running' to false.
if the number of nodes is small like 500 per request, it works fine.

Zvicii, 2020-04-01:
I have met the same problem.
I build a client and a server using open62541.
but when i try to create monitoreditems using UA_Client_MonitoredItems_createDataChanges, if the number is too large(5000 string nodeid), it gets stuck and doesnt return.the server would also stuck in its loop and cant be sopped by setting 'running' to false.
if the number of nodes is small like 500 per request, it works fine.
@jpfr