open62541/open62541 #2948: Completion of error handling

elfring, 2019-07-31:
Would you like to add more error handling for return values from functions like the following?
- [malloc](https://pubs.opengroup.org/onlinepubs/9699919799/functions/malloc.html "Memory allocation") ⇒ [connection_send](https://github.com/open62541/open62541/blob/347a486f538ffe7c076e6e6e1dd0d2cd578efeff/arch/network_ws.c#L67)
- [pthread_mutex_init](https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_mutex_init.html "Initialize a mutex.") ⇒ [UA_WorkQueue_init](https://github.com/open62541/open62541/blob/6aa4d8d9d2f76b0ba95acba06345ae38c3e4c068/src/ua_workqueue.c#L19)

Pro, 2019-07-31:
Yes. Every malloc call should be followed by a Null Check. And wherever possible return values should be checked and handled correspondingly.

jpfr, 2019-07-31:
Hmm. CI didn't trigger on the websocket connection.
Should be activated for clang-analyser.