open62541/open62541 #2949: Monitoring objects which are source of event subscriptions

sqadasi, 2019-07-31:
## Description
Hello everyone,
I have a question regarding monitoring objects which are source of event subscriptions (A source of event subscriptions is an Object that has its “SubscribeToEvents” bit set within the EventNotifier Attribute).
Using the open62541 events example and UaExpert client, I was able to monitor **all** objects in the event view. Is this behavior correct? I would've expected that only objects with their EventNotifier Attribute set to "SubscribeToEvent" could be monitored for events?

Also I would like to ask a more general question regarding the HasEventSource and EventNotifier. Are those references mandatory in the hierarchy of Event Objects? At the moment the hierarchy of Event Objects is browsed recursively using HasComponent and Organizes references only:
https://github.com/open62541/open62541/blob/9095278e9fdd00f367f6d24790c2b63005e8cda8/src/server/ua_subscription_events.c#L376

Shouldn't we be using also HasEventSource and EventNotifier references to determine whether we allowed to get Event Notification while monitoring a specific Object? An exception here is the Server Object, which is when monitored for events, must get all Event Notifications implying the Server Object has HasEventSource references to all event source nodes.

please see: Part 3, 7.17 and 7.18

@aDogCalledSpot , @jpfr 

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [X] open62541 Version (master):
 - [X] Other OPC UA SDKs used (client or server): Client: UaExpert
 - [X] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-07-31:
These are good points.
We should double-check the conditions for event propagation.

And put a big comment at the top of `ua_subscription_events.c` that describes **all conditions**

- When an event can be generated
- How an event is propagated

@sqadasi Can you just paste the relevant quotes from the relevant sections for the standard?

I would prefer if we had the big comment first. And then correct the conditions accordingly.

sqadasi, 2019-08-01:
@jpfr, thank you for responding!
Although the selected quotes are a bit too large, they shall provide a thorough description of the hierarchical structure of EventNotifiers. However, how an event can be generated depends strongly on the selected filter (I guess), which is not really discussed in the selected quotes (I need more research here).

- Part 3, 4.6.1:
"Events represent specific transient occurrences. System configuration changes and system
errors are examples of Events. Event Notifications report the occurrence of an Event. Events
defined in this document are not directly visible in the OPC UA AddressSpace. **Objects and
Views can be used to subscribe to Events. The EventNotifier Attribute of those Nodes identifies
if the Node allows subscribing to Events. Clients subscribe to such Nodes to receive
Notifications of Event occurrences.**
Event Subscriptions use the Monitoring and Subscription Services defined in Part 4 to subscribe
to the Event Notifications of a Node.
**Any OPC UA Server that supports eventing shall expose at least one Node as EventNotifier. The Server Object defined in Part 5 is used for this purpose. Events generated by the Server are available via this Server Object.** A Server is not expected to produce Events if the connection to the event **source is down for some reason (i.e. the system is offline).
Events may also be exposed through other Nodes anywhere in the AddressSpace. These Nodes (identified via the EventNotifier Attribute) provide some subset of the Events generated by the Server. The position in the AddressSpace dictates what this subset will be**. For example, a process area Object representing a functional area of the process would provide Events originating from that area of the process only. It should be noted that this is only an example and it is fully up to the Server to determine what Events should be provided by which Node."

- Part 3, 7.17:
"The HasEventSource ReferenceType is a concrete ReferenceType and can be used directly. It is a subtype of HierarchicalReferences.
The semantic of this ReferenceType is to relate event sources in a hierarchical, non-looping organization. **This ReferenceType and any subtypes are intended to be used for discovery of Event generation in a Server.** **They are not required to be present for a Server to generate an Event from its source (causing the Event) to its notifying Nodes. In particular, the root notifier of a Server, the Server Object defined in Part 5, is always capable of supplying all Events from a Server and as such has implied HasEventSource References to every event source in a Server.**
**The SourceNode of this ReferenceType shall be an Object that is a source of event subscriptions. A source of event subscriptions is an Object that has its “SubscribeToEvents” bit set within the EventNotifier Attribute.
The TargetNode of this ReferenceType can be a Node of any NodeClass that can generate event notifications via a subscription to the reference source.**
Starting from Node “A” and only following References of the HasEventSource ReferenceType
or of its subtypes it shall never be possible to return to “A”. But it is permitted that, following the
References, there may be more than one path leading to another Node “B”."

- Part 3, 7.18:
"The HasNotifier ReferenceType is a concrete ReferenceType and can be used directly. It is a
subtype of HasEventSource.
**The semantic of this ReferenceType is to relate Object Nodes that are notifiers with other
notifier Object Nodes. The ReferenceType is used to establish a hierarchical organization of
event notifying Objects.** It is a subtype of the HasEventSource ReferenceType defined in 7.16.
**The SourceNode of this ReferenceType shall be Objects or Views that are a source of event
subscriptions. The TargetNode of this ReferenceType shall be Objects that are a source of
event subscriptions. A source of event subscriptions is an Object that has its
“SubscribeToEvents” bit set within the EventNotifier Attribute.**
If the TargetNode of a Reference of this type generates an Event, then this Event shall also be
provided in the SourceNode of the Reference.
An example of a possible organization of Event References is represented in Figure 27. In this
example an unfiltered Event subscription directed to the “Pump” Object will provide the Event
sources “Start” and “Stop” to the subscriber. An unfiltered Event subscription directed to the
“Area 1” Object will provide Event sources from “Machine B”, “Tank A” and all notifier sources
below “Tank A”."
[image]

- Part 4, 5.12.1.1:
"Objects and views can be used to monitor Events. Events are only available from Nodes where the
SubscribeToEvents bit of the EventNotifier Attribute is set."

- Part 5, 8.3.2:
"The Server Object serves as root notifier, that is, its EventNotifier Attribute shall be set providing
Events. All Events of the Server shall be accessible subscribing to the Events of the Server
Object."

- Also, it could be helpful to provide some quotes from "OPC Uinified Architecture, by Mahnke, Leitner and Damm, 2009 Springer-Verlag", 5.5.2.6:
"A second tree control can be populated with EventNotifiers and Event sources provided by the server. **The References used to build such a hierarchy are Organizes, HasNotifier, and HasEventSource. Starting points for such hierarchies are the Server Object, the Objects Folder, or Views.** Table 5.24 describes the Browse parameters used to fill a tree control with the EventNotifier and Event source hierarchy.
**The EventNotifiers are used to create Monitored Items which is a preselection of Events provided by this EventNotifier. The Event sources can be used as filter criteria.**"

[image]

jojastahl, 2019-10-30:
This is releated to, maybe even a duplicate of, #2474

gimmyy, 2020-06-19:
hi! in the description, you mentioned "please see: Part 3, 7.17 and 7.18", could you please tell me what is it? @sqadasi

jojastahl, 2020-06-22:
Taken from OPC UA Part 3 - Address Space Model Release 1.04 Specification:

>#### 7.17 Has Event Source
>... This ReferenceType and any subtypes are intended to be used for discovery of Event generation in a Server. They are not required to be present for a Server to generate an Event from its source (causing the Event) to its notifying Nodes. In particular, the root notifier of a Server, the Server Object defined in Part 5, is always capable of supplying all Events from a Server and as such has implied HasEventSource References to every event source in a Server.
>
>The SourceNode of this ReferenceType shall be an Object that is a source of event subscriptions. ...
>
>The TargetNode of this ReferenceType can be a Node of any NodeClass that can generate event notifications via a subscription to the reference source.
>...
>
>#### 7.18 HasNotifier
>...
>The semantic of this ReferenceType is to relate Object Nodes that are notifiers with other
notifier Object Nodes. The ReferenceType is used to establish a hierarchical organization of
event notifying Objects. It is a subtype of the HasEventSource ReferenceType defined in 7.16.
>...
>If the TargetNode of a Reference of this type generates an Event, then this Event shall also be
provided in the SourceNode of the Reference.
>
>An example of a possible organization of Event References is represented in Figure 27. In this
example an unfiltered Event subscription directed to the “Pump” Object will provide the Event
sources “Start” and “Stop” to the subscriber. An unfiltered Event subscription directed to the
“Area 1” Object will provide Event sources from “Machine B”, “Tank A” and all notifier sources
below “Tank A”.
>
>[image]
> ...

jpfr, 2020-06-23:
Was this solved via this commit last year?
https://github.com/open62541/open62541/commit/e6baba149f4f66d3ddb40baacb91354e2d812509#diff-dd7eb0316f0b90c03ec8a6123b429054

sqadasi, 2020-06-25:
> 
> 
> Was this solved via this commit last year?
> [e6baba1#diff-dd7eb0316f0b90c03ec8a6123b429054](https://github.com/open62541/open62541/commit/e6baba149f4f66d3ddb40baacb91354e2d812509#diff-dd7eb0316f0b90c03ec8a6123b429054)

I would say only partly (I guess further commits have optimized the solution as well). 
Server Object emits events correctly and the hierarchy of event sources seems to include event sources only once (without repitition), so we don't get an event for each reference between the monitored item and the event source. 
However, monitoring an object which has its “SubscribeToEvents” bit not set within the EventNotifier Attribute is still possible. Creating such a monitored item should be rejected by the server.

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue and describe it so clearly.

The problem has since been fixed on the maintained branches (see #3918, d8be4ac and e6baba1), so we are closing it as resolved.

closed 2026-10-20 by o6-nbeier