open62541/open62541 #2951: Reading value attribute with UA_TYPES_STRING type

pierrekurkdjian, 2019-07-31:
I am asynchronously reading and writing value of OPCUA nodes using respectively UA_Client_writeValueAttribute_async() UA_Client_readValueAttribute_async(). 
This works fine with all the supported data types (UA_TYPES_STRING, UA_TYPES_BOOLEAN, UA_TYPES_DOUBLE, etc) as long as the nodes have the root node as parent node. 

When the node is not directly related to the root node, I cannot read the string-typed value of the node (segmentation fault). No problem with all the other typed (UA_TYPES_BOOLEAN, UA_TYPES_DOUBLE, etc). 

Example: I have a root node called "MAIN". MAIN has 2 kids: "MyNodeString1" and "MyNodeStruct". MyNodeStruct has 1 kid called "MyNodeString2". 
Now I try to read the value attribute of the 2 nodes (UA_TYPES_STRING). 

MAIN.MyNodeString1.value (no problem, I can read it)
MAIN.MyNodeStruct.MyNodeString2.value (segmentation fault)

Is there someone else having the same issue?

jpfr, 2019-07-31:
Hmm. I don't think the parent of the node is a likely root cause.
That would be very surprising.

Can you reproduce the problem in a debugger and send (a screenshot of) the backtrace?

Just to understand: Is the server or the client receiving a segfault?

pierrekurkdjian, 2019-07-31:
Thank you very much for your very fast answer. 
The segmentation fault is on the client side. 

First I send the request by calling UA_Client_readValueAttribute_async: 
void Node::read_value(std::string node_name){
  UA_StatusCode retval = UA_Client_readValueAttribute_async(
    client, 
    UA_NODEID_STRING(node_name_space, const_cast<char*>(node_name.c_str())), 
    &Node::receive_callback_aux, 
    this, 
    &reqId
  );
}

Then I receive the answer some milliseconds later and receive_callback() is called: 
void Node::receive_callback(UA_Client *client, void *userdata, UA_UInt32 requestId, UA_Variant *var){
      std::cout << "answer received"  << std::endl;
      if(data_type == UA_TYPES_STRING) {
        UA_String uastr = *(UA_String*) var->data;
        char str[OPCUA_STRING_MAX_LENGTH];
        strcpy(str, (char*)uastr.data);
        std::cout << "node value = " << str  << " /" << std::endl;
      }
}

I read the 2 values: 
n.read_value("MAIN.MyNodeString1");
n.read_value("MAIN.MyNodeStruct.MyNodeString2");

Execution: 
answer received
node value = blabla of MyNodeString1 /
Segmentation fault (core dumped)
                OR sometimes
answer received
node value = blabla of MyNodeString1 /
answer received
Segmentation fault (core dumped)

The error doesn't occur at the same point so I imagine the error happens in another thread created by the library. 
If I replace "MAIN.MyNodeStruct.MyNodeString2" by a fake name (a non existing node name) then I receive no answer (and no error) of course. 

GDB doesn't give me more information.

jpfr, 2019-08-01:
Don't use strcpy but strncpy instead.
UA_String is not NULL-terminated.
It instead comes with a separete `length` field.

closed 2019-08-01 by jpfr

pierrekurkdjian, 2019-08-01:
In fact I had this issue when the string value I read is an empty string. 
Thanks so much