open62541/open62541 #2952: Support for Push authentication

epistax, 2019-07-31:
Hello all,

We're are investigating support of OPC-UA for a very big customer. One of their requirements is push authentication. From the UA spec chapter 12 section 7.3, I see that the very first part of push authentication is support for the CreateSigningRequest command, which doesn't appear present today. We're potentially prepared to contribute to open62541 to fill the gap, but I need to understand the scope of the gap so I can compare against alternatives and make a proposal to my group. 

What is missing from where open62541 is today? What other questions should I maybe be asking?

Thank you!

jpfr, 2019-08-01:
What you describe is the Global Discovery Server (GDS) feature.

We have a first implementation of the GDS. But it is not ready for production.
https://github.com/open62541/open62541/pull/2299

Development on the GDS will pick up again later this year. 
If you want to be in the loop, write me at julius.pfrommer@iosb.fraunhofer.de .

epistax, 2019-08-01:
Thank you @jpfr. I couldn't tell if the GDS feature support meant for the creation of a GDS, or for the interaction with a GDS. From your comment, it sounds like it's talking about the interaction with GDS. Or I'm mis-reading the UA spec (which could be the case). Do I have this correct?

jpfr, 2019-08-01:
You can use the GDS with any OPC UA client.
But in addition, you need features to swap out the client certificate at runtime, and so on.

closed 2020-12-06 by jpfr