open62541/open62541 #2959: Is there any example to monitor alarm content

fanucwj, 2019-08-02:
hello
i have contact with Siemens 828d device,but alarm content is not available,can i obtain it through Monitor method? expect to have example 
thank you!

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

schroeder-, 2019-08-02:
Alarms for the client are like normale events. See: open62541/examples/tutorial_client_events.c .
On top you can call 
`UA_StatusCode retval = UA_Client_call(client, UA_NODEID_NUMERIC(0, UA_NS0ID_CONDITIONTYPE), UA_NODEID_NUMERIC(0, UA_NS0ID_CONDITIONTYPE_CONDITIONREFRESH), 1, *inp, &outputSize, &output);` 
after you montiored a event, to get all current active alarms in your monitored item callback.

fanucwj, 2019-08-05:
Thank you for you replying @schroeder- 
but i run the open62541/examples/tutorial_client_events.c ,it seems to run error: malloc(): memory corruption: 0x0001fea0 ***
any other example to references?

closed 2019-08-07 by fanucwj