open62541/open62541 #2963: Cannot extend Events

winks, 2019-08-02:
## Description
Trying to use `examples/tutorial_server_events` + examples/tutorial_client_events` with added fields leads to segfault.
I may very well be using it wrong, but the docs are not detailed enough

## Background Information / Reproduction Steps

```
./tutorial_server_events
./tutorial_client_events
# call "Generate Event" via UaExpert (this works)
```

```
# output on the server:
[2019-08-02 19:36:49.630 (UTC+0200)] info/userland	Creating event
[2019-08-02 19:36:49.631 (UTC+0200)] error/userland	Alarms and Conditions are not supported yet!
[2019-08-02 19:36:49.723 (UTC+0200)] info/network	Connection 7 | Closed

# output on the client:
[2019-08-02 19:36:49.723 (UTC+0200)] info/userland	Notification
[2019-08-02 19:36:49.723 (UTC+0200)] info/userland	Message: 'An event has been generated.'
[2019-08-02 19:36:49.723 (UTC+0200)] info/userland	Severity: 100
zsh: segmentation fault  ./bin/tutorial_client_events opc.tcp://localhost:4840
```
--- /opt/src/open62541-v1.0-rc1/examples/tutorial_server_events.c	2019-08-02 19:27:16.762038961 +0200
+++ tutorial_server_events.c	2019-08-02 19:27:47.302439175 +0200
@@ -1,9 +1,7 @@
 /* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
  * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */
 
-#include <open62541/plugin/log_stdout.h>
-#include <open62541/server.h>
-#include <open62541/server_config_default.h>
+#include <open62541/open62541.h>
 
 #include <signal.h>
 #include <stdlib.h>
@@ -75,6 +73,10 @@
     UA_Server_writeObjectProperty_scalar(server, *outId, UA_QUALIFIEDNAME(0, "SourceName"),
                                          &eventSourceName, &UA_TYPES[UA_TYPES_STRING]);
 
+    UA_UInt32 answer = 42;
+	UA_Server_writeObjectProperty_scalar(server, *outId, UA_QUALIFIEDNAME(0, "Answer"), 
+                                         &answer, &UA_TYPES[UA_TYPES_UINT32]);
+
     return UA_STATUSCODE_GOOD;
 }
```

```
--- /opt/src/open62541-v1.0-rc1/examples/tutorial_client_events.c	2019-07-31 16:25:37.143907310 +0200
+++ tutorial_client_events.c	2019-08-02 19:34:04.187377419 +0200
@@ -1,13 +1,7 @@
 /* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
  * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */
 
-#include <open62541/client.h>
-#include <open62541/client_config_default.h>
-#include <open62541/client_highlevel.h>
-#include <open62541/client_subscriptions.h>
-#include <open62541/plugin/log_stdout.h>
-#include <open62541/server.h>
-#include <open62541/server_config_default.h>
+#include <open62541/open62541.h>
 
 #include <signal.h>
 
@@ -44,6 +38,9 @@
         if(UA_Variant_hasScalarType(&eventFields[i], &UA_TYPES[UA_TYPES_UINT16])) {
             UA_UInt16 severity = *(UA_UInt16 *)eventFields[i].data;
             UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Severity: %u", severity);
+        } else if(UA_Variant_hasScalarType(&eventFields[i], &UA_TYPES[UA_TYPES_UINT32])) {
+            UA_UInt32 answer = *(UA_UInt32 *)eventFields[i].data;
+            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Answer: %u", answer);
         } else if (UA_Variant_hasScalarType(&eventFields[i], &UA_TYPES[UA_TYPES_LOCALIZEDTEXT])) {
             UA_LocalizedText *lt = (UA_LocalizedText *)eventFields[i].data;
             UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
@@ -56,7 +53,7 @@
     }
 }
 
-const size_t nSelectClauses = 2;
+const size_t nSelectClauses = 3;
 
 static UA_SimpleAttributeOperand *
 setupSelectClauses(void) {
@@ -91,6 +88,18 @@
     selectClauses[1].attributeId = UA_ATTRIBUTEID_VALUE;
     selectClauses[1].browsePath[0] = UA_QUALIFIEDNAME_ALLOC(0, "Severity");
 
+    selectClauses[2].typeDefinitionId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEEVENTTYPE);
+    selectClauses[2].browsePathSize = 1;
+    selectClauses[2].browsePath = (UA_QualifiedName*)
+        UA_Array_new(selectClauses[1].browsePathSize, &UA_TYPES[UA_TYPES_QUALIFIEDNAME]);
+    if(!selectClauses[2].browsePath) {
+        UA_SimpleAttributeOperand_deleteMembers(selectClauses);
+        UA_SimpleAttributeOperand_delete(selectClauses);
+        return NULL;
+    }
+    selectClauses[2].attributeId = UA_ATTRIBUTEID_VALUE;
+    selectClauses[2].browsePath[0] = UA_QUALIFIEDNAME_ALLOC(0, "Answer");
+
     return selectClauses;
 }

```

Used CMake options:

```bash
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON ..                            

```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0-rc1
 - [x] Other OPC UA SDKs used (client or server): none
 - [x] Operating system: Ubuntu 18.04
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

sqadasi, 2019-08-05:
Did you extend BaseEventType first with own EventType as SubType which holds "Answer" field with HasProperty or HasComponent reference to the EventType? Your new field should also have the ModellingRule reference "mandatory", so the field would be created when creating the event object instance.
I hope it helps somehow...

winks, 2019-08-05:
@sqadasi 
Thanks for your answer, that helped quite a bit.

For now I used this snippet from `tutorial_server_object.c` and it seems to basically work.

```
    UA_VariableAttributes mnAttr = UA_VariableAttributes_default;
    mnAttr.displayName = UA_LOCALIZEDTEXT("en-US", "ManufacturerName");

    currentPos_ = UA_STRING("Pump King Ltd.");
    UA_Variant_setScalar(&mnAttr.value, &currentPos_, &UA_TYPES[UA_TYPES_STRING]);

    UA_Server_addVariableNode(server, UA_NODEID_NULL, eventType,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "ManufacturerName"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), mnAttr, NULL, &manufacturerNameId_);
    /* Make the manufacturer name mandatory */
    UA_Server_addReference(server, manufacturerNameId_,
                           UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
                           UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY), true);
```

aDogCalledSpot, 2019-09-16:
Closing since it seems this issue is solved, please reopen if it is not.

closed 2019-09-16 by aDogCalledSpot