open62541/open62541 #2965: Security Policy

ywchen1994, 2019-08-05:
##Security Policy

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -BUILD_SHARED_LIBS=ON
cmake -UA_ENABLE_ENCRYPTION=ON

```

Following is my error
SecurityPolicy not specified-> use default #None
error/client Failed to find the required security policy
error/client couldn't connect the client to a TCP secure channel

Following is my code
https://drive.google.com/open?id=1xOY9xVn19hHJFPj68geie-w-4Sp9ErVx

I created key.der and cert.der by using \tools\certs\create_self-signed.py 

Here is my command
sudo python create_self-signed.py /home/user/file
I got both *.der successfully,but I still can't connect to siemens 840D controller.
## Checklist

Please provide the following information:

 - [*] open62541 Version (release number or git tag):master
 - [*] Other OPC UA SDKs used (client or server): uaExpert
 - [*] Operating system:ubuntu16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [*] Self-contained code example attached
 - [*] Critical issue

mlgiraud, 2019-08-06:
Is it intentional that you are initializing both policies to the first array element? (i.e. [0])

ywchen1994, 2019-08-06:
I had trid different type,but it did'nt work.

schroeder-, 2019-08-06:
I think your cert is not set or wrong. I had similar issues as you discripe, when i could't load my cert.

mlgiraud, 2019-08-06:
Could you supply a wireshark dump?

ywchen1994, 2019-08-08:
@schroeder- Could you give me more detail about creating cert？I tried key and cert which are in 828D controller.
@mlgiraud  wireshark is only support Windows,isn't it ? Eventhough I use wireshark,what kind of situation should I use, capturing while using uaExpert or capturing while using my project？(p.s. My project was built in Linux)

schroeder-, 2019-08-08:
First thing to check is if the cert files contain any data after loadFile. 
Second is to raise the log level and check the log file. Because if your certs are invalid it will be shown in the log file. 
Also you should check the return value of UA_SecurityPolicy_Basic128Rsa15 and other functions.

ywchen1994, 2019-08-09:
I think my cert and key are correct,because I add return status from Base128 to check is UA_STATUSCODE_GOOD or not .Then it return UA_STATUSCODE_GOOD.But it still failed ,while connecting  to server.The following is the error.
error/client	Could not activate the Session with StatusCode BadInternalError

ywchen1994, 2019-08-13:
@schroeder- Could you tell me how to use create_self-signed.py ,I think I miss the trustlist step

xcvb85, 2019-08-14:
@ywchen1994: I used the following command to create the certificate:
> ./create_self-signed.py .

I tried to connect to the Siemens control with the "access_control_encrypt" example (only changed username and password).

This is the console output:
> [2019-08-14 15:49:36.025 (UTC+0200)] info/client	Connecting to endpoint opc.tcp://192.168.213.16:4840
[2019-08-14 15:49:36.025 (UTC+0200)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-08-14 15:49:36.025 (UTC+0200)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-08-14 15:49:36.025 (UTC+0200)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2019-08-14 15:49:36.025 (UTC+0200)] debug/client	Initialize the SecurityPolicy context
[2019-08-14 15:49:36.026 (UTC+0200)] info/client	SecurityPolicy not specified -> use default #None
[2019-08-14 15:49:36.026 (UTC+0200)] warn/securitypolicy	Security policy None is used to create SecureChannel. Accepting all certificates
[2019-08-14 15:49:36.051 (UTC+0200)] info/client	TCP connection established
[2019-08-14 15:49:36.051 (UTC+0200)] debug/network	Sent HEL message
[2019-08-14 15:49:36.076 (UTC+0200)] debug/network	Received ACK message
[2019-08-14 15:49:36.076 (UTC+0200)] debug/channel	Requesting to open a SecureChannel
[2019-08-14 15:49:36.076 (UTC+0200)] debug/channel	OPN message sent
[2019-08-14 15:49:36.105 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 0 | Decrypting chunk
[2019-08-14 15:49:36.105 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 0 | Chunk size before and after decryption: 136, 136
[2019-08-14 15:49:36.106 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 0 | Verifying chunk signature
[2019-08-14 15:49:36.106 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 0 | Decrypted and verified chunk with request id 1 and sequence number 51
[2019-08-14 15:49:36.106 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 0 | Sequence Number processed: 51
[2019-08-14 15:49:36.106 (UTC+0200)] debug/client	Decode a message of type OpenSecureChannelResponse
[2019-08-14 15:49:36.106 (UTC+0200)] info/client	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2019-08-14 15:49:36.106 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 109964 | Generating new local keys
[2019-08-14 15:49:36.106 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 109964 | Generating new remote keys
[2019-08-14 15:49:36.106 (UTC+0200)] info/client	Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2019-08-14 15:49:36.106 (UTC+0200)] debug/client	Sending a request of type 426
[2019-08-14 15:49:36.132 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 109964 | Decrypting chunk
[2019-08-14 15:49:36.132 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 109964 | Chunk size before and after decryption: 4227, 4227
[2019-08-14 15:49:36.132 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 109964 | Decrypted and verified chunk with request id 2 and sequence number 52
[2019-08-14 15:49:36.132 (UTC+0200)] trace/channel	Connection 3 | SecureChannel 109964 | Sequence Number processed: 52
[2019-08-14 15:49:36.132 (UTC+0200)] debug/client	Decode a message of type GetEndpointsResponse
[2019-08-14 15:49:36.132 (UTC+0200)] info/client	Found 3 endpoints
[2019-08-14 15:49:36.132 (UTC+0200)] info/client	Rejecting endpoint 0: security mode doesn't match
[2019-08-14 15:49:36.132 (UTC+0200)] info/client	Endpoint 1 has 2 user token policies
[2019-08-14 15:49:36.132 (UTC+0200)] info/client	Rejecting UserTokenPolicy 0 (anonymous) in endpoint 1: configuration doesn't match
[2019-08-14 15:49:36.132 (UTC+0200)] info/client	Selected Endpoint opc.tcp://192.168.213.16:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-08-14 15:49:36.132 (UTC+0200)] info/client	Selected UserTokenPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15 with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-08-14 15:49:36.132 (UTC+0200)] info/client	Disconnect to switch to a different SecurityPolicy
[2019-08-14 15:49:36.132 (UTC+0200)] info/client	Connecting to endpoint opc.tcp://192.168.213.16:4840
[2019-08-14 15:49:36.133 (UTC+0200)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-08-14 15:49:36.133 (UTC+0200)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-08-14 15:49:36.133 (UTC+0200)] warn/client	The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2019-08-14 15:49:36.133 (UTC+0200)] debug/client	Initialize the SecurityPolicy context
[2019-08-14 15:49:36.158 (UTC+0200)] info/client	TCP connection established
[2019-08-14 15:49:36.158 (UTC+0200)] debug/network	Sent HEL message
[... cut ...]
[2019-08-14 15:49:36.266 (UTC+0200)] debug/securitypolicy	Deleted members of EndpointContext for sp_basic128rsa15
[2019-08-14 15:49:36.266 (UTC+0200)] debug/securitypolicy	Deleted members of EndpointContext for sp_basic256
[2019-08-14 15:49:36.266 (UTC+0200)] debug/securitypolicy	Deleted members of EndpointContext for sp_basic256sha256

In Wireshark I see that I have received the following error message from the controller:
> Error: 0x80140000 [BadCertificateTimeInvalid]

Not sure what I do wrong.

teccheck, 2025-02-25:
There hasn't been any activity for quite a long time and the code has changed quite a lot during this time, so I will close this issue

closed 2025-02-25 by teccheck