open62541/open62541 #2967: Does Pub/Sub performance depends on the UDP performance?

kingalexsylvester, 2019-08-05:
Hello,

1. Can we conclude that Pub/Sub performance solely depends upon the performance of the UDP stack of the system!?

2. So if I want to achieve better real time performance, my first step should be to optimize UDP performance than to modify Pub/Sub stack?

3. For a reliable system, the end user must be cautious in the way he is choosing his TCP/IP stack or else he will not be able to achieve high performance. Is my understanding right?

4. I am planning to do some measurements on my system to understand how much does UDP contributes and how much does the Publisher code contributes for a single publisher packet. If someone has the details please share it.

5. Will moving to L2 Publisher be a wiser decision?(so this can eliminate the need of IP stack and my sensor can run only the publisher code and never depend on TCP/IP is it technically possible)

jpfr, 2019-08-23:
1. Not only. But this is the first bottleneck you will encounter.

2. Your first step should be to measure the duration. And then work on the actual bottleneck **in your system**. The bottleneck will be in different places for different systems.

3. This why we are working on TSN for deterministic communication.

4. You can take some inspiration for taking time-measurements here:
https://github.com/open62541/open62541/blob/master/tests/server/check_server_readspeed.c

5. What is L2?

existedinnettw, 2024-02-02:
I think it mean layer 2 in OSI model (data-link communication layer). 
Based on my known, OPC UA doesn't support any protocol that can run without IP stack.

teccheck, 2025-02-25:
The given answers should be enough. Feel free to reopen this issue if needed

closed 2025-02-25 by teccheck