open62541/open62541 #2979: Callbacks not called on server Historic  API

juangburgos, 2019-08-08:
Hello open62541 guys,

I have been trying to understand how to use the historic API to implement a custom backend. I am starting simple by trying to re-create the in-memory example but in C++ using Qt.

I saw some guys attempted to do [something similar before](https://github.com/open62541/open62541/issues/2396) and promised to post the results, but never did.

Any ways, in my tests, I have noticed that not all callbacks of the API are called when changing value or requesting historical data form UaExpert. Specifically, in order, callbacks are called as follows:

### When Value Changed

* serverSetHistoryData

### On Historic Request

* boundSupported
* timestampsToReturnSupported
* getEnd
* firstIndex
* lastIndex
* getDateTimeMatch
* resultSize
* copyDataValues

But all the other callbacks are not called (`getDataValue`, `insertDataValue`, `updateDataValue`, etc.).

My question is; How can I create a request (using UaExpert, or client code) to trigger the call of the remaining callbacks?

I would like to know and understand what need to be put in there, by means of debugging.

Below I post my test code.

## Test Code

`main.cpp`

```c++
#define NOMINMAX

#include <QCoreApplication>
#include <QTimer>
#include <QDebug>
#include <QHash>
#include <QMap>

#include <algorithm>
#include <iterator>

#include <open62541.h>

typedef UA_NodeId UA_DataBaseKey;
typedef QMap<UA_DateTime, UA_DataValue> UA_DateValueMap;

uint qHash(const UA_DataBaseKey &key)
{
	return UA_NodeId_hash(&key);
};
bool operator==(const UA_DataBaseKey &e1, const UA_DataBaseKey &e2)
{
	return e1.namespaceIndex == e2.namespaceIndex
		&& e1.identifierType == e2.identifierType
		&& (e1.identifierType == UA_NODEIDTYPE_NUMERIC    ? e1.identifier.numeric == e2.identifier.numeric :
			e1.identifierType == UA_NODEIDTYPE_STRING     ? UA_String_equal    (&e1.identifier.string    , &e2.identifier.string    ) :
			e1.identifierType == UA_NODEIDTYPE_GUID       ? UA_Guid_equal      (&e1.identifier.guid      , &e2.identifier.guid      ) :
			e1.identifierType == UA_NODEIDTYPE_BYTESTRING ? UA_ByteString_equal(&e1.identifier.byteString, &e2.identifier.byteString) : false);
}

QHash<UA_DataBaseKey, UA_DateValueMap> m_database;

UA_HistoryDataBackend MyBackend() {
	UA_HistoryDataBackend result;
[... 429 lines cut ...]

TARGET = 10_xtest_server_historicaldata_qt
CONFIG += console
CONFIG -= app_bundle

TEMPLATE = app

INCLUDEPATH += $$PWD/

include($$PWD/../../src/amalgamation/open62541.pri)

INCLUDEPATH += $$PWD/../../depends/open62541.git/include
INCLUDEPATH += $$PWD/../../depends/open62541.git/plugins/include

# NOTE : compile amalgamation with UA_ENABLE_HISTORIZING

SOURCES += main.cpp
```

## Checklist

 - [x] open62541 Version (release number or git tag): master
 - [x] Other OPC UA SDKs used (client or server): UaExpert
 - [x] Operating system: Windows
 - [x] Self-contained code example attached

peter-rustler-basyskom, 2019-08-08:
Hi @juangburgos 
If i remember right, there was no way to update and insert historical data by UA-Expert at the time of writing historical data server interface.
So, I had to use the client side of historical data in open62541 interface to test this things.

You can have a look in this test for example:
https://github.com/open62541/open62541/blob/master/tests/client/check_client_historical_data.c
This uses the client api side of things.
You could write a small test client which fits your needs.

You also can have a look into:
https://github.com/open62541/open62541/blob/master/tests/server/check_server_historical_data.c
There you do find some example which calls the service method directly. This is some kind of "dirty", but helps with debugging to some degree.

For getDataValue you need to request the data so, that the clauses will become true at this place:
https://github.com/open62541/open62541/blob/6aa4d8d9d2f76b0ba95acba06345ae38c3e4c068/plugins/historydata/ua_history_database_default.c#L268
LLONG_MIN has a special meaning there which stand for TIMESTAMP_UNSPECIFIED.

If you have values present in the database and start timestamp of the client request is TIMESTAMP_UNSPECIFIED aka LLONG_MIN for example would do the trick.

I hope that helps.

juangburgos, 2019-08-08:
It does help. Thank you very much. I was not aware of those tests. So now I have more material to cover. I believe historic access is a key feature for OPC UA servers.

I hope in the near feature provide a Qt C++ API for historic access to [QUaServer](https://github.com/juangburgos/QUaServer). Let me know if you have any ideas on how you would design a nice easy-to-use API for C++.

Thanks.

closed 2019-08-08 by juangburgos