open62541/open62541 #2989: How to identify DataSetMessage if Undefined_0 ordering is used?

techstein111, 2019-08-12:
By opc-ua pubsub specifications, `DataSetWriterId` in `DataSetMessage` header is optional, and `DataSetMessage` ordering can be set to `Undefined_0`, which is when we can't guarantee an order of DataSetMessages in NetworkMessage. How to identify which `DataSetMessage` we received if there are multiple ones in the `WriterGroup`?

andreasebner, 2019-08-13:
According to the standard, one option should be the use of 'DataSetMetaData Message (UADP)' 1.04 - 7.2.2.4.2.4 and 'DataSetWriter Configuration Message (UADP)' 1.04 - 7.2.2.4.2.5). These messages include e.g. an array of DataSetWriterIds which are contained in the configuration information. Unfortunately, we do not support these messages as well as the ordering of DataSetMessages. We are currently working on the meta data topic and will extend the PubSub implementation soon.

velimirj, 2019-08-13:
So {{DataSetWriterId}} would not be optional if we want to identify {{DataSetMessage}} when there are more than one in {{WriterGroup}}?

techstein111, 2019-08-13:
So ``DataSetWriterId`` would not be optional in case we want to identify ``DataSetMessage`` in ``WriterGroup`` that contains more than one?

andreasebner, 2019-08-14:
Yes, we produce (currently) the dsm internally based on the order within the WriterGroup but your code should not depend on this implementation detail.

o6-nbeier, 2026-10-20:
Thank you for the question.

With undefined ordering, identify each DataSetMessage by the DataSetWriterIds in the NetworkMessage PayloadHeader (the order is not a stable identity), or use a fixed preconfigured layout if you omit that header.

Closing for now.

closed 2026-10-20 by o6-nbeier