open62541/open62541 #2992: AddNodes: Node could not create a node with error code BadNodeAttributesInvalid

parth-ifam, 2019-08-13:
## Starting a server with the nodeset `Opc.Ua.Nodeset2.xml` and `myNS.xml` from Example

Compiled the `myNS.h`and `myNS.c` successfully. Having problems to start the server with it.
Without `myNS` it works.
```
[2019-08-13 16:24:28.820 (UTC+0200)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2019-08-13 16:24:28.820 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Node could not create a node with error code BadNodeAttributesInvalid
[2019-08-13 16:24:28.820 (UTC+0200)] error/server	Could not add the example nodeset. Check previous output for any error.

```
 Did I forget any step in between ?

## Checklist

Please provide the following information:

 - [x] open62541 Version (v1.0-rc4):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

parth-ifam, 2019-08-13:
Update: Issue solved by adding `-DUA_ENABLE_AMALGAMATION` while compiling.  
```gcc -std=c99 -DUA_ENABLE_AMALGAMATION open62541.c  main.c myNS.c -o main```
 Dont know if this is the right way.

Jeffrey-zhang313, 2021-03-17:
Actually, there is a correspondence between header files open62541.h and your source code. Can fix this problem by update open62541.h when config is changed

SeanMWX, 2022-11-08:
Did you solve this problem? I got this problem in v1.1, with "AddNodes: Node could not create a node with error code BadNodeAttributesInvalid"

prrw, 2022-11-08:
Yes it was solved. This is quite an old issue. See my comment before. Seems you have some different error.

SeanMWX, 2022-11-08:
> Yes it was solved. This is quite an old issue. See my comment before. Seems you have some different error.

Hello, thanks for your reply!

I just the solution [here](https://github.com/open62541/open62541/issues/4927#issuecomment-1077896892). Not sure the reason, but when I changed my Ubuntu from 22.04 to 18.04. This error disappears and everything works well.

My version:
v1.1 + Ubuntu 18.04

prrw, 2022-11-08:
i think i had it working with Ubuntu 20.04

teccheck, 2025-02-25:
It seems to be working now. I will close this issue. Feel free to reopen if needed

closed 2025-02-25 by teccheck