open62541/open62541 #2997: Method Call - Callback method with XML

parth-ifam, 2019-08-14:
@jpfr How is it possible to register a method callback when your method node is defined as per  `myNS.xml` in `myNS.c`. How to pass the reference of the callback for the method ? 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0-rc4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-08-15:
https://open62541.org/doc/current/server.html#method-callbacks

closed 2019-08-15 by jpfr

parth-ifam, 2019-08-15:
Thanks.. works now