open62541/open62541 #3000: Build Issue (Typo?)

juangburgos, 2019-08-14:
# Description

Cannot build latest amalgamation on Windows10 MSVC 2017, because it cannot find a file that does exists but the path seems to have a typo.

## Background Information / Reproduction Steps

I get the following error:

```
9>Integrating file 'C:/Users/Pepe/Desktop/REPOS/QUaServer.git/depends/open62541.git/plugins/ua_accesscontrol_default.c'...done.
9>Integrating file 'C:/Users/Pepe/Desktop/REPOS/QUaServer.git/depends/open62541.git/plugins/ua_pki_default.c'...done.
9>Integrating file 'C:/Users/Pepe/Desktop/REPOS/QUaServer.git/depends/open62541.git/plugins/ua_nodestore_default.c'...done.
9>Integrating file 'C:/Users/Pepe/Desktop/REPOS/QUaServer.git/depends/open62541.git/plugins/ua_config_default.c'...done.
9>Integrating file 'C:/Users/Pepe/Desktop/REPOS/QUaServer.git/depends/open62541.git/plugins/securityPolicies/ua_securitypolicy_none.c'...done.
9>Traceback (most recent call last):
9>  File "C:/Users/Pepe/Desktop/REPOS/QUaServer.git/depends/open62541.git/tools/amalgamate.py", line 69, in <module>
9>    with io.open(fname, encoding='utf8', errors='replace') as infile:
9>FileNotFoundError: [Errno 2] No such file or directory: 'C:/Users/Pepe/Desktop/REPOS/QUaServer.git/depends/open62541.git/plugins/securityPolcies/securitypolicy_mbedtls_common.c'
9>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(209,5): error MSB6006: "cmd.exe" exited with code 1.
9>Done building project "open62541-amalgamation-source.vcxproj" -- FAILED.
```

But the file does exist:

[image]

It seems the error is because in the path  `/open62541.git/plugins/securityPolcies/securitypolicy_mbedtls_common.c`, the **securityPolcies** is mispelled.

Used CMake options:

```bash
cmake ./.. -DUA_ENABLE_AMALGAMATION=ON -G "Visual Studio 15 2017"
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master. 1.0rc4
 - [x] Operating system: Windows
 - [x] Self-contained code example attached
 - [x] Critical issue

juangburgos, 2019-09-10:
I fount out this onle occurs on one machine. Every time I try to build on that machine it eats out letter of paths for some reason I cannot explaint. So closing issue, since it must be due to the environment.

closed 2019-09-10 by juangburgos

stegrofan1jo, 2020-12-15:
Any news on this one? I am having the same issue. Can someone giv an explanation how to track this down? I think its an issue of path length (location in file tree) or some encoding issue of line endings during git checkout.  I am struggeling to debug the amalgamation script.

vprasolov, 2022-07-05:
I have same issue with loosing letter in some of paths. When location was changed another letter in other path is missing. I’m using vcpkg manager on Windows 10 machine with VS 2022 installed.

Daen-cpu, 2025-03-29:
same issue (trying to install qt6 with vcpkg, windows 10, vs2022 msvc), smth new?