open62541/open62541 #3001: Git-based version detection fails for zip download

jpfr, 2019-08-15:
The git-based version detection will fail if the user downloads a zip file off github.

@Pro

Pro, 2019-08-15:
There was a more detailed discussion about this in #2980 

As mentioned there it is not the best approach to download the zip directly. That one for example also doesn't include the submodules. It's better to git clone or use our other prepared distributions.

If one downloads the zip, he can still set the version manually using the `OPEN62541_VERSION` Option for cmake.

If we do not want to use the automatic detection, we have to manually pish the version in CMake, Debian, Arch and other package distributions. As we have seen for the 1.0-RC4 release, that is very likely to fail. With the automatic detection every package's git commit and base tag is automatically set.

If one really needs to download the zip source, there is the automatically updated `pack/master` and `pack/1.0` which already has the correct version set and the submodules inlined.

I'd go for the automatic detection since it saves us a lot of work. IMHO that's the more error-safe way.

If really necessary, we can switch back to the old hardcoded way.
@jpfr any strong opinion?

jpfr, 2019-08-16:
> If one really needs to download the zip source, there is the automatically updated pack/master and pack/1.0 which already has the correct version set and the submodules inlined.

This is pretty cool. But it needs to be documented better.
And we need links to the zip file for these branches on our homepage and and the readme.

We do this for the users. So they should know it exists.

Pro, 2019-08-16:
> But it needs to be documented better.
> And we need links to the zip file for these branches on our homepage and the readme.

Yes! Full ACK.
It's already on my TODO. I first wanted to see how the automatic detection works out. There's still a bug for the pack branches: If you create a tag, travis doesn't push them. Will fix it in the coming days.

PR #3061 (2019-08-30): fix(docs): Add info about pack branches
fixes #3001
commit 34412b14 (2019-08-30): fix(docs): Add info about pack branches

fixes #3001
closed 2019-09-02 by Pro