open62541/open62541 #3002: Connecting with basic256sha256 to prosys server

EirikAlbrektsen, 2019-08-15:
## Description
I am trying to connect an open62541 client to a Prosys OPCUA server using encryption and preferably user login. In this effort, I have made a client that is able to connect using basic128rsa15 and basic256 security policy with both anonymous and username/password as UserTokenPolicy working for both security policies. So I know that my certificates and all that is setup correctly. However when I try to use basic256sha256 as a security policy, which the Prosys server supports, it is only able to connect using anynomous UserTokenPolicy. The only difference between the three test runs is which security policy I am using, otherwise everything is identical.  Is there something special about the basic256sha256 policy that I am missing? 

I am using Windows 10 to run Prosys server, with a wsl Ubuntu 18.10 running the open62541 client.

See below for code, here are the three logs: 
### BASIC256 with Username LOG: 
```
/mnt/c/Users/eirik/Documents/RocketFarmAS/OPCUATest/problemClient
[2019-08-15 13:01:36.721 (UTC+0200)] info/client	Connecting to endpoint opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer
[2019-08-15 13:01:36.721 (UTC+0200)] debug/client	Initialize the SecurityPolicy context
[2019-08-15 13:01:36.721 (UTC+0200)] info/client	SecurityPolicy not specified -> use default #None
[2019-08-15 13:01:36.721 (UTC+0200)] warn/securitypolicy	Security policy None is used to create SecureChannel. Accepting all certificates
[2019-08-15 13:01:36.721 (UTC+0200)] info/client	TCP connection established
[2019-08-15 13:01:36.721 (UTC+0200)] debug/network	Sent HEL message
[2019-08-15 13:01:36.724 (UTC+0200)] debug/network	Received ACK message
[2019-08-15 13:01:36.725 (UTC+0200)] debug/channel	Requesting to open a SecureChannel
[2019-08-15 13:01:36.725 (UTC+0200)] debug/channel	OPN message sent
[2019-08-15 13:01:36.725 (UTC+0200)] debug/client	Decode a message of type OpenSecureChannelResponse
[2019-08-15 13:01:36.726 (UTC+0200)] info/client	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2019-08-15 13:01:36.726 (UTC+0200)] info/client	Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2019-08-15 13:01:36.726 (UTC+0200)] debug/client	Sending a request of type 426
[2019-08-15 13:01:36.735 (UTC+0200)] debug/client	Decode a message of type GetEndpointsResponse
[2019-08-15 13:01:36.735 (UTC+0200)] info/client	Found 4 endpoints
[2019-08-15 13:01:36.735 (UTC+0200)] info/client	Endpoint 0 has 7 user token policies
[2019-08-15 13:01:36.735 (UTC+0200)] info/client	Rejecting UserTokenPolicy 0 (anonymous) in endpoint 0: configuration doesn't match
[2019-08-15 13:01:36.735 (UTC+0200)] info/client	Rejecting UserTokenPolicy 1 (token) in endpoint 0: configuration doesn't match
[2019-08-15 13:01:36.735 (UTC+0200)] info/client	Selected Endpoint opc.tcp://DESKTOP-L2EDEO6:53530/OPCUA/SimulationServer with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-08-15 13:01:36.735 (UTC+0200)] info/client	Selected UserTokenPolicy username_basic256 with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-08-15 13:01:36.735 (UTC+0200)] info/client	Disconnect to switch to a different SecurityPolicy
[2019-08-15 13:01:36.735 (UTC+0200)] info/client	Connecting to endpoint opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer
[2019-08-15 13:01:36.735 (UTC+0200)] debug/client	Initialize the SecurityPolicy context
[2019-08-15 13:01:36.736 (UTC+0200)] info/client	TCP connection established
[2019-08-15 13:01:36.736 (UTC+0200)] debug/network	Sent HEL message
[2019-08-15 13:01:36.739 (UTC+0200)] debug/network	Received ACK message
[2019-08-15 13:01:36.740 (UTC+0200)] debug/channel	Requesting to open a SecureChannel
[2019-08-15 13:01:36.744 (UTC+0200)] debug/channel	OPN message sent
[2019-08-15 13:01:36.877 (UTC+0200)] debug/client	Decode a message of type OpenSecureChannelResponse
[2019-08-15 13:01:36.877 (UTC+0200)] info/client	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2019-08-15 13:01:36.877 (UTC+0200)] debug/client	Create a new session
[2019-08-15 13:01:36.877 (UTC+0200)] debug/client	Sending a request of type 459
[2019-08-15 13:01:36.958 (UTC+0200)] debug/client	Decode a message of type CreateSessionResponse
[2019-08-15 13:01:36.960 (UTC+0200)] debug/client	Sending a request of type 465
[2019-08-15 13:01:37.010 (UTC+0200)] debug/client	Decode a message of type ActivateSessionResponse
Program End
[2019-08-15 13:01:37.010 (UTC+0200)] debug/client	Sending a request of type 471
[2019-08-15 13:01:37.026 (UTC+0200)] debug/client	Decode a message of type CloseSessionResponse
[2019-08-15 13:01:37.026 (UTC+0200)] debug/securitypolicy	Deleted members of EndpointContext for sp_basic256
```
### BASIC256sha256 with anynoumous LOG: 
```
/mnt/c/Users/eirik/Documents/RocketFarmAS/OPCUATest/problemClient
[2019-08-15 13:31:41.045 (UTC+0200)] info/client	Connecting to endpoint opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer
[2019-08-15 13:31:41.045 (UTC+0200)] debug/client	Initialize the SecurityPolicy context
[2019-08-15 13:31:41.045 (UTC+0200)] info/client	SecurityPolicy not specified -> use default #None
[2019-08-15 13:31:41.045 (UTC+0200)] warn/securitypolicy	Security policy None is used to create SecureChannel. Accepting all certificates
[2019-08-15 13:31:41.046 (UTC+0200)] info/client	TCP connection established
[2019-08-15 13:31:41.046 (UTC+0200)] debug/network	Sent HEL message
[2019-08-15 13:31:41.049 (UTC+0200)] debug/network	Received ACK message
[2019-08-15 13:31:41.049 (UTC+0200)] debug/channel	Requesting to open a SecureChannel
[2019-08-15 13:31:41.049 (UTC+0200)] debug/channel	OPN message sent
[2019-08-15 13:31:41.050 (UTC+0200)] debug/client	Decode a message of type OpenSecureChannelResponse
[2019-08-15 13:31:41.050 (UTC+0200)] info/client	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2019-08-15 13:31:41.050 (UTC+0200)] info/client	Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2019-08-15 13:31:41.050 (UTC+0200)] debug/client	Sending a request of type 426
[2019-08-15 13:31:41.058 (UTC+0200)] debug/client	Decode a message of type GetEndpointsResponse
[2019-08-15 13:31:41.058 (UTC+0200)] info/client	Found 4 endpoints
[2019-08-15 13:31:41.058 (UTC+0200)] info/client	Rejecting endpoint 0: security policy doesn't match
[2019-08-15 13:31:41.058 (UTC+0200)] info/client	Endpoint 1 has 7 user token policies
[2019-08-15 13:31:41.058 (UTC+0200)] info/client	Selected Endpoint opc.tcp://DESKTOP-L2EDEO6:53530/OPCUA/SimulationServer with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[... 162 lines cut ...]
    for(size_t deleteCount = 0; deleteCount < revocationListSize; deleteCount++) {
        UA_ByteString_clear(&revocationList[deleteCount]);
    }
    UA_Client_delete(client); /* Disconnects the client internally */
}

```

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> -BUILD_SHARED_LIBS=OFF -.UA_ENABLEENCRYPTION= ON -DEBUGLEVEL=200 ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

schroeder-, 2019-08-15:
Did you tried to use UAExpert to connect with the Basic256sha256 and username?

EirikAlbrektsen, 2019-08-15:
I have tried to use the UAExpert client to connect to the Prosys server with those yes and that worked fine

EirikAlbrektsen, 2019-09-04:
Fixed the problem, the solution was to add either basic256 or basic128rsa15 as a security policy as well. My endpoint supported all three security policies for the channel, but only allowed for the two other policies for userTokenPolicy which is what determines the authentication policy. This is set automatically by open62541 as long as one of the two are available to the client you want to use.
````/mnt/c/Users/eirik/Documents/RocketFarmAS/OPCUATest/cppClient
LibOpen62541: successfully configured with UA_SecurityPolicy_None
LibOpen62541: successfully configured with UA_SecurityPolicy_Basic256Sha256
LibOpen62541: successfully configured with UA_SecurityPolicy_Basic128Rsa15
[2019-09-04 08:40:50.109 (UTC+0200)] info/client	Connecting to endpoint opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer
[2019-09-04 08:40:50.109 (UTC+0200)] info/client	SecurityPolicy not specified -> use default #None
[2019-09-04 08:40:50.109 (UTC+0200)] warn/securitypolicy	Security policy None is used to create SecureChannel. Accepting all certificates
[2019-09-04 08:40:50.110 (UTC+0200)] info/client	TCP connection established
[2019-09-04 08:40:50.114 (UTC+0200)] info/client	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2019-09-04 08:40:50.114 (UTC+0200)] info/client	Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Found 1 endpoints
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Endpoint 0 has 4 user token policies
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Rejecting UserTokenPolicy 0 (certificate) in endpoint 0: configuration doesn't match
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Rejecting UserTokenPolicy 2 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Selected Endpoint opc.tcp://DESKTOP-L2EDEO6:53530/OPCUA/SimulationServer with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Selected UserTokenPolicy username_basic128 with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Disconnect to switch to a different SecurityPolicy
[2019-09-04 08:40:50.117 (UTC+0200)] info/client	Connecting to endpoint opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer
[2019-09-04 08:40:50.118 (UTC+0200)] info/client	TCP connection established
[2019-09-04 08:40:50.222 (UTC+0200)] info/client	Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
Value of the read attribute: -1.732051 
Browsing nodes in objects folder:
NAMESPACE NODEID           BROWSE NAME      DISPLAY NAME    
5                          "Eirik"          Simulation/"Eirik"
5                          Counter1         Simulation/Counter1
5                          Expression1      Simulation/Expression1
5                          Random1          Simulation/Random1
5                          Sawtooth1        Simulation/Sawtooth1
5                          Sinusoid1        Simulation/Sinusoid1
5                          Sinusoid2        Simulation/Sinusoid2
5                          Square1          Simulation/Square1
5                          Triangle1        Simulation/Triangle1
Program End
````

closed 2019-09-04 by EirikAlbrektsen

I-Yoo, 2020-04-21:
Could you help me solve this problem: #3563