open62541/open62541 #3010: Chosen NodeId

miehe-dup, 2019-08-19:
## Description

Follow-up to #2873 -- similar: A solution here might pose a solution to #2873, but not vice versa.

## Background Information / Reproduction Steps

In #2873, an (implicitly) proposed workaround is `UA_ENABLE_DETERMINISTIC_RNG` CMake option. This allows for persistent NodeIds -- but we may want some logic behind them.

We can choose (or at least, request) the resulting NodeId using the API: `UA_Server_addXNode(server, desiredNodeId, ...);` 

However, when adding a Node type containing children, these children's NodeIds are randomized.

> * Can I change the NodeId of a node at runtime?
> * Can I set the NodeId of an automatically instantiated child node? … (Obviously) only relevant for nodes created from objecttypes or variabletypes, e.g. refer to tutorial_server_object.c

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): master
 - [X] Other OPC UA SDKs used (client or server): None
 - [X] Operating system: Any
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-08-19:
Does #3014 solve the issue?

miehe-dup, 2019-08-20:
> Does #3014 solve the issue?

Looks very promising, to say the least :-)

jpfr, 2019-08-24:
#3014 was merged. Closing this.
Please reopen if required.

closed 2019-08-24 by jpfr