open62541/open62541 #3011: Rust bindings

miehe-dup, 2019-08-19:
## Description

I started some work on Rust bindings for open62541 at https://github.com/miehe-dup/open62541-sys. It compiles the library, but my "simple" example does not work.

I am aware of https://github.com/locka99/opcua as an alternative, yet it lacks key features, e.g. the nodeset compiler. In the meantime, it might be viable to create working FFI bindings for Rust.

PRs and discussions are welcome! :)

jpfr, 2019-08-19:
Nice! Rust is the future. The future just isn't evenly distributed. :-)
Feel free to add this to the wiki.
We have a list of open source OPC UA implementations and a list of bindings to open62541.

konradsz, 2022-02-11:
For anyone interested in using open62541 in Rust via FFI - I was able to compile and run two examples programs: https://github.com/konradsz/open62541-ffi

mati865, 2024-02-27:
There is a new project providing low and high level bindings but currently only for the client; https://github.com/HMIProject/open62541

o6-nbeier, 2026-10-20:
Since there is no recent activity and this is no issue of open62541 itself, we will close this for now.

closed 2026-10-20 by o6-nbeier