open62541/open62541 #3015: 1.0-rc4 build failure with ARM target

GrantEdwards, 2019-08-19:
I've been using 0.2.0 and 0.3.0 for some time and am now trying to move to 1.0-rc4,
but I have run into some build failures with arm-linux-gcc 4.6.3.

Invalid command line options passed to compiler:

  `-Wpedantic`  is unrecognized
  `-Wno-static-inline` is unrecognized
 
Code in ua_securechannel.c that jumps over a variable initialization:

    src/ua_securechannel.c: In function 'sendSymmetricChunk':
    src/ua_securechannel.c:782:9: error: jump skips variable initialization [-Werror=jump-misses-init]
    src/ua_securechannel.c:821:1: note: label 'error' defined here
    src/ua_securechannel.c:792:12: note: 'total_length' declared here
    src/ua_securechannel.c:782:9: error: jump skips variable initialization [-Werror=jump-misses-init]
    src/ua_securechannel.c:821:1: note: label 'error' defined here
    src/ua_securechannel.c:790:12: note: 'pre_sig_length' declared here

Variable in ua_server_ns0.c that shadows a global from <sys/socket.h>:

    src/server/ua_server_ns0.c: In function 'readStatus':
    src/server/ua_server_ns0.c:283:20: error: declaration of 'shutdown' shadows a global declaration [-Werror=shadow]

Code in ua_services_attribute.c that compares an unsigned value (a  UA_TimestampsToReturn enum variable) with 0:

    src/server/ua_services_attribute.c: In function 'Service_Read':
    src/server/ua_services_attribute.c:403:5: error: comparison of unsigned expression < 0 is always false [-Werror=type-limits]

Variable in ua_subscription_datachange.c that shadows a global from <strings.h>

    src/server/ua_subscription_datachange.c: In function 'outOfPercentDeadBand':
    src/server/ua_subscription_datachange.c:79:73: error: declaration of 'index' shadows a global declaration [-Werror=shadow]

Variable in  ua_securitypolicy_none.c that shadows a global from <stdlib.h>

    plugins/securityPolicies/ua_securitypolicy_none.c: In function 'generateNonce_none':
    plugins/securityPolicies/ua_securitypolicy_none.c:79:19: error: declaration of 'rand' shadows a global declaration [-Werror=shadow]
    plugins/securityPolicies/ua_securitypolicy_none.c:85:15: error: declaration of 'rand' shadows a global declaration [-Werror=shadow]

----
Here's my cmake command line:

    cmake -DUA_ENABLE_ENCRYPTION=ON -DMBEDTLS_INCLUDE_DIRS=/home/nextgen/opcua-open62541-update/mbedtls-2.16.2/include -DMBEDTLS_LIBRARY=/home/nextgen/opcua-open62541-update/mbedtls-2.16.2/library/libmbedtls.a -DMBEDCRYPTO_LIBRARY=/home/nextgen/opcua-open62541-update/mbedtls-2.16.2/library/libmbedcrypto.a -DMBEDX509_LIBRARY=/home/nextgen/opcua-open62541-update/mbedtls-2.16.2/library/libmbedx509.a -DCMAKE_TOOLCHAIN_FILE=../Toolchain-arm-linux.cmake ../open62541

And here's the toolchain file:

    # this one is important
    SET(CMAKE_SYSTEM_NAME Linux)
    #this one not so much
    SET(CMAKE_SYSTEM_VERSION 1)

    # specify the cross compiler
    SET(CMAKE_C_COMPILER   /home/nextgen/toolchain/bin/arm-linux-gcc)
    SET(CMAKE_CXX_COMPILER /home/nextgen/toolchain/bin/arm-linux-g++)

    # where is the target environment 
    SET(CMAKE_FIND_ROOT_PATH  /home/nextgen/toolchain/arm-unknown-linux-uclibcgnueabi)

    # search for programs in the build host directories
    SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
    # for libraries and headers in the target directories
    SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
    SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

GrantEdwards, 2019-08-19:
Attached is the patch I'm currently using to get a clean build.  I'm _not_ suggesting this 
patch be used, but I just wanted to provide a starting point.

I can't think of any way to avoid conditionally removing -Wpedantic and
-Wno-static-in-inline options, since they're unrecognized by the compiler.

However, removing the -Wshadow as shown in the patch is probably less
desirable than fixing the C code to avoid shadowing globals.

I'm undecided on whether adding the -Wno-jump-misses-init flag is a good idea
or if the code should just be fixed to avoid that one.

[arm-4.6.3-build.patch.txt](https://github.com/open62541/open62541/files/3517439/arm-4.6.3-build.patch.txt)

GrantEdwards, 2019-08-19:
I don't understand why, but It appears that removing the -Wpedantic flag 
also eliminates the error caused by the unrecognized -Wno-static-in-inline
option:

With both -Wpedantic and -Wno-static-in-inline:

    cc1: error: unrecognized command line option '-Wpedantic'
    cc1: error: unrecognized command line option "-Wno-static-in-inline" [-Werror]
    cc1: all warnings being treated as errors

But removing only -Wpedantic eliminates _both_ errors.

jpfr, 2019-08-19:
Have you tried with a more recent gcc?
Errors should be similar to the architectures we test against.

Anyway, we will incoporate the improvements where we can.

GrantEdwards, 2019-08-20:
At the moment, I don't have a more recent arm-linux-gcc available, but I would guess most of the above errors are gcc version-dependent rather than architecture dependent.  The one that might be architecture dependent is the unsigned value compared with 0 error.  It's possible that enums are unsigned on ARM and signed on x86.

If you would find it helpful, I could build a newer arm cross-compiler and try it.  However, using that newer compiler to build production firmware isn't an option at this time because it would require a large investment in re-testing and re-certifying products.

That said, I don't see how we can enable -Wshadow, include <stdlib.h> and <string.h>, and then declare a local variables named 'rand' and 'index' without expecting warnings/errors.  I realize that some versions of gcc complain when a variable shadows another variable but not when a variable shadows a function, but I don't think it's good to rely on that.

Would it be appropriate to use cmake's check_c_compiler_flag() to determine if -Wpedantic is supported by the compiler?

jpfr, 2020-07-07:
We have added a CMake macro that tests compiler flags before adding them.
Should be resolved with the latest release. If problems remain in your setup, please reopen.

closed 2020-07-07 by jpfr