open62541/open62541 #3021: Questionable usage of add_definitions() in CMake code

DerDakon, 2019-08-21:
## Description

```add_definitions()``` is for definitions, that is ```-Dsomething```. Most of the things that are passed into it are instead compiler flags (e.g. ```-Wsomething```), which would belong into CMAKE_&lt;lang&gt;_FLAGS*.

## Background Information / Reproduction Steps

I tried to clean this up, but there are more things to consider which I lack the project knowledge:

 * I personally would update to the requirement to at least CMake 3.1, which could then use ```set(CMAKE_C_STANDARD 99)``` and get rid of ```-std=c99```. This option is removed in [tests/fuzz/CMakeLists.txt](https://github.com/open62541/open62541/blob/master/tests/fuzz/CMakeLists.txt#L28) again. Why?
 * definitions apply to any language, while the compiler flag variables are language specific. The flags would need to be set to both C and CXX if they should appear there, while I suspect that there are also flags that do only work in one mode.
 * I guess some of these flags do not work on all compilers or versions, see #3015 for example.
 * UA_ARCH_REMOVE_FLAGS could cause trouble, as it will not work then anymore. The way out for a user would be to provide it's own set of compiler flags and entirely override the automatically set ones (i.e. don't provide a build type).
 * I wonder if many of these flags should only affect debug modes, and not release.

mlgiraud, 2019-08-22:
If we are modifying this anyways, maybe we could use [target_compile_options](https://cmake.org/cmake/help/latest/command/target_compile_options.html) and the other corresponding `target_` commands.

DerDakon, 2019-08-22:
This could raise the bar for the needed CMake version even more, depending on which of those commands you want to use.

mlgiraud, 2019-08-23:
afaik all the target commands are available since version 3.0

closed 2025-02-25 by teccheck

crbn13, 2025-11-06:
Which recent versions is this fixed in? because i just spent the last day trying to work out why my cmake project kept failing to build, and it turns out it was because things being set in [CMakeLists.txt](https://github.com/open62541/open62541/blob/master/CMakeLists.txt) line 595 were being applied to a **different libraries** compile arguments and causing that other libraries build to fail. Both libraries were included in the CMake project using add_subdirectory()

open62541/CMakeLists.txt:

    if((CMAKE_COMPILER_IS_GNUCC OR CMAKE_C_COMPILER_ID STREQUAL "Clang") AND
    NOT ${UA_ARCHITECTURE} STREQUAL "zephyr")
        check_add_cc_flag("-std=c99")   # C99 mode
        check_add_cc_flag("-pipe")      # Avoid writing temporary files (for compiler speed)
        check_add_cc_flag("-Wall")      # Warnings
        check_add_cc_flag("-Wextra")    # More warnings
        check_add_cc_flag("-Wpedantic") # Standard compliance
    

specifically the -std=c99 being applied was causing the issue and i fixed it with the following

    
        # check_add_cc_flag("-std=c99")   # C99 mode
        set(C_STANDARD 99)
        set(C_STANDARD_REQUIRED True)