open62541/open62541 #3027: variableNodes fail to be instantiated

Darker7, 2019-08-22:
## Description

Whenever I try to add a VariableNode to the server it fails with BadNodeAttributesInvalid.

## Background Information / Reproduction Steps

I'm currently going through the introduction to open62541 to learn about OPC UA and of course that includes trying out and playing with the tutorial code. Problem is, it doesn't work, so either the tutorial code is out of date (though it seems to be compliant with the model as provided in the docs), there's a problem with the source code and version 1.0 is broken (but why didn't anybody raise a fuss about this earlier then?) or something strange is going on on my end (but my system should be too standard for that to happen).

Used CMake options:

Used ccmake to turn on UA_ENABLE_AMALGAMATION and UA_ENABLE_SUBSCRIPTIONS_EVENTS.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0
 - [x] Other OPC UA SDKs used (client or server): none
 - [x] Operating system: Linux (Ubuntu 18.04 in a VirtualBox on a Windows 10 host)
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

## Logs

```bash
darker@playground:~/tutorials$ ./addingVariablesToAServer 
[2019-08-22 11:53:35.498 (UTC+0200)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2019-08-22 11:53:35.498 (UTC+0200)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Node could not create a node with error code BadNodeAttributesInvalid
[2019-08-22 11:53:35.498 (UTC+0200)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Node could not create a node with error code BadNodeAttributesInvalid
Writing a string returned statuscode BadNodeIdUnknown
```

jpfr, 2019-08-23:
Can you post the source code where you add the variable?
Impossible to tell otherwise.

Darker7, 2019-08-26:
Well, it's the tutorial code as I said, but if you want a direct copy of what I have, here: https://paste2.org/48HJ3Ut5

Darker7, 2019-08-26:
Found the solution by accident. The problem was that the copied tutorial code had the includes un-amalgamated but asked for an amalgamated open62541.h which I complied with and this ended up causing problems; now I replaced the includes with one `#include "open62541.h"` and it works as intended.

Here's the corresponding paragraph from the tutorials:  
[page](https://open62541.org/doc/current/tutorial_server_firststeps.html)

> To get started, downdload the open62541 single-file release from http://open62541.org or generate it according to the *[build instructions](https://open62541.org/doc/current/building.html#building)* with the “amalgamation” option enabled. From now on, we assume you have the `open62541.c/.h` files in the current folder. Now create a new C source-file called `myServer.c` with the following content:

```C
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

static volatile UA_Boolean running = true;
```

closed 2019-08-26 by Darker7

EugenM-sie, 2020-11-20:
@Darker7 Do you find Solution?