open62541/open62541 #3028: Event server methods definitions are missing

Darker7, 2019-08-22:
## Description

UA_Server_createEvent and UA_Server_triggerEvent are not defined.

## Background Information / Reproduction Steps

I'm currently going through the introduction to open62541 to learn about OPC UA and of course that includes trying out and playing with the tutorial code. Problem is, it doesn't work, so either there's a problem with the source code and version 1.0 is broken (but why didn't anybody raise a fuss about this earlier then?), the ~~tutorial~~example code is out of date or something strange is going on on my end (but my system should be too standard for that to happen).  
Well, or I'm just failing to find the events with UaExpert.

Used CMake options:

Used ccmake to turn on UA_ENABLE_AMALGAMATION and UA_ENABLE_SUBSCRIPTIONS_EVENTS.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0
 - [x] Other OPC UA SDKs used (client or server): none
 - [x] Operating system: Linux (Ubuntu 18.04 in a VirtualBox on a Windows 10 host)
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

## Compiler message

```bash
darker@playground:~/tutorials$ gcc -std=c99 open62541.c eventServer.c -o eventServer
eventServer.c: In function ‘setUpEvent’:
eventServer.c:53:28: warning: implicit declaration of function ‘UA_Server_createEvent’; did you mean ‘UA_Server_browseNext’? [-Wimplicit-function-declaration]
     UA_StatusCode retval = UA_Server_createEvent(server, eventType, outId);
                            ^~~~~~~~~~~~~~~~~~~~~
                            UA_Server_browseNext
eventServer.c: In function ‘generateEventMethodCallback’:
eventServer.c:107:14: warning: implicit declaration of function ‘UA_Server_triggerEvent’; did you mean ‘UA_Server_browseNext’? [-Wimplicit-function-declaration]
     retval = UA_Server_triggerEvent(server, eventNodeId,
              ^~~~~~~~~~~~~~~~~~~~~~
              UA_Server_browseNext
```

schroeder-, 2019-08-22:
You have to also set UA_ENABLE_SUBSCRIPTIONS in cmake

Darker7, 2019-08-22:
UA_ENABLE_SUBSCRIPTIONS is on by default.

jpfr, 2019-08-22:
Of course it is defined in the server.h.
See here:
https://github.com/open62541/open62541/blob/be4330cc402477c1cd378084a50d3b7eb30160e7/include/open62541/server.h#L1276

Maybe you are on version 0.3 of the SDK...

Darker7, 2019-08-22:
Ah, now I see the problem. There is a clash of definitions within tutorial code because it calls the standard <open62541/server.h> which has UA_ENABLE_SUBSCRIPTIONS_EVENTS turned off but is supposed to be compiled with the custom open62541.c which includes the custom open62541.h which has events enabled.

So the solution is: modify the tutorial code's `#includes` to use "open62541.h" instead of <open62541/???.h>

jpfr, 2019-08-22:
There is no standard server.h
It contains whatever features have been enabled in cmake.

closed 2019-08-22 by jpfr

betodr4, 2020-11-13:
Hi, could you please explain me how to modify Cmake Options. I am using Eclipse sothere I can only modify Menuconfig paameters.

I am developing an OPCUA project and I would like to use events and alarms on it. I am just using open62541.c in the project, but I still have problems because I must not include server.h or other files.

Please help me to solve that.
Thank you very much.