open62541/open62541 #3029: bug with a method call with a custom datatype argument

Todovski, 2019-08-22:
## Description

Hi,

I need to call with a client (Ua_expert or other OPC_UA client) a method  with a custom datatype parameter and get an int32 back,
all the calls of methods without custom datatypes work well against which I put a custom dataype in parameter I have a fatal error when my server decodes the extension object.
My custom dataype are well initialized at startup, I can write and read these variables without problems.
My problem is just the method call 

the bug :
[image]

 UA_DataTypeArray *customTypes points to a wrong address...

I tried with different datatypes but the result is the same

see my struct : 
typedef struct {
    UA_Int32 varInt32;
    UA_Float varFloat;
    UA_Double varDouble;
    UA_Byte varByte;
} UA_STR_DataType;

See my methode declaration with the generate_datatypes
`
/* m3 - ns=1;i=56 */

static UA_StatusCode function_Model_23_begin(UA_Server *server, UA_UInt16* ns) {
#ifdef UA_ENABLE_METHODCALLS
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_MethodAttributes attr = UA_MethodAttributes_default;
attr.displayName = UA_LOCALIZEDTEXT("", "m3");
#ifdef UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS
attr.description = UA_LOCALIZEDTEXT("", "m3");
#endif
attr.writeMask = 0;
attr.userWriteMask = 0;
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_METHOD,
UA_NODEID_NUMERIC(ns[1], 56),
UA_NODEID_NUMERIC(ns[1], 49),
UA_NODEID_NUMERIC(ns[0], 47),
UA_QUALIFIEDNAME(ns[0], "m3"),
 UA_NODEID_NULL,
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_METHODATTRIBUTES],NULL, NULL);
return retVal;
#else
return UA_STATUSCODE_GOOD;
#endif /* UA_ENABLE_METHODCALLS */
}

static UA_StatusCode function_Model_23_finish(UA_Server *server, UA_UInt16* ns) {
#ifdef UA_ENABLE_METHODCALLS
return UA_Server_addMethodNode_finish(server, 
UA_NODEID_NUMERIC(ns[1], 56)
, NULL, 0, NULL, 0, NULL);
#else
return UA_STATUSCODE_GOOD;
#endif /* UA_ENABLE_METHODCALLS */
}

/* InputArguments - ns=1;i=57 */

static UA_StatusCode function_Model_24_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.minimumSamplingInterval = 10.000000;
attr.userAccessLevel = 3;
attr.accessLevel = 3;
attr.valueRank = 1;
attr.arrayDimensionsSize = 1;
UA_UInt32 arrayDimensions[1];
arrayDimensions[0] = 0;
attr.arrayDimensions = &arrayDimensions[0];
attr.dataType = UA_NODEID_NUMERIC(ns[0], 296);
UA_Argument variablenode_ns_1_i_57_variant_DataContents[1];

UA_init(&variablenode_ns_1_i_57_variant_DataContents[0], &UA_TYPES[UA_TYPES_ARGUMENT]);
variablenode_ns_1_i_57_variant_DataContents[0].name = UA_STRING("vStr");
variablenode_ns_1_i_57_variant_DataContents[0].dataType = UA_NODEID_NUMERIC(ns[1], 22);
variablenode_ns_1_i_57_variant_DataContents[0].valueRank = (UA_Int32) -1;
UA_Variant_setArray(&attr.value, &variablenode_ns_1_i_57_variant_DataContents, (UA_Int32) 1, &UA_TYPES[UA_TYPES_ARGUMENT]);
attr.displayName = UA_LOCALIZEDTEXT("", "InputArguments");
#ifdef UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS
attr.description = UA_LOCALIZEDTEXT("", "InputArguments");
#endif
attr.writeMask = 0;
attr.userWriteMask = 0;
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_VARIABLE,
UA_NODEID_NUMERIC(ns[1], 57),
UA_NODEID_NUMERIC(ns[1], 56),
UA_NODEID_NUMERIC(ns[0], 46),
UA_QUALIFIEDNAME(ns[0], "InputArguments"),
UA_NODEID_NUMERIC(ns[0], 68),
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_VARIABLEATTRIBUTES],NULL, NULL);

return retVal;
}

static UA_StatusCode function_Model_24_finish(UA_Server *server, UA_UInt16* ns) {
return UA_Server_addNode_finish(server, 
UA_NODEID_NUMERIC(ns[1], 57)
);
}`

Do you have an idea of the problem?
Thank you for your comeback

## Background Information / Reproduction Steps

Used CMake options:

Commandline options:
-DUA_ENABLE_AMALGAMATION:BOOL="1" -DBIBER_COMPILER:FILEPATH="BIBER_COMPILER-NOTFOUND" 

Cache file:
UA_ENABLE_AMALGAMATION:BOOL=1
BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
Include all CMake options here, which you modified or used for your build.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (1.0-rc1 and 1.0-rc5-0):
 - [ ] Other OPC UA SDKs used ( server + Ua_expert): 
 - [ ] Operating system: w7 x64

starki93, 2019-09-18:
Did you tried to the the extension object like this:

` auto ua_arg = (UA_Argument *) object.content.decoded.data;`

teccheck, 2025-02-25:
The code path has changed quite a lot since this issue was opened. It should be fixed now. Feel free to reopen if this issue persists

closed 2025-02-25 by teccheck