open62541/open62541 #3031: Trying to learn

AngerlFG, 2019-08-22:
I'm making a little project with a sensor and raspberry Pi as a server and I have learned the basis for opc ua. Now I'd like to make a little code to collect data from that sensor. But from the examples in the documentation, I've noticed that some functions are not defined.[Like UA_LOG_INFO();] So, how can I know what it does?
In other forum, someone said to me to look into the source code, where can i find it?What should I do know?
Sorry if I'm not being very clear. I have very little knowledge programming.
Thank you

miehe-dup, 2019-08-23:
I am using `UA_LOG_*` functions too, they are defined in `<open62541/plugin/log_stdout.h>`. Do you use CMake?

AngerlFG, 2019-08-23:
For building the library. I don't understand a thing xD, is this code?
```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information.
 *
 *    Copyright 2016, 2018 (c) Fraunhofer IOSB (Author: Julius Pfrommer)
 */

#ifndef UA_LOG_STDOUT_H_
#define UA_LOG_STDOUT_H_

#include <open62541/plugin/log.h>

_UA_BEGIN_DECLS

extern UA_EXPORT const UA_Logger UA_Log_Stdout_; /* Logger structure */
extern UA_EXPORT const UA_Logger *UA_Log_Stdout; /* Shorthand pointer */

/* Don't use these definitions. They are only exported as long as the client
 * config is static and required compile-time  */
UA_EXPORT void
UA_Log_Stdout_log(void *_, UA_LogLevel level, UA_LogCategory category,
                  const char *msg, va_list args);
UA_EXPORT void
UA_Log_Stdout_clear(void *logContext);

_UA_END_DECLS

#endif /* UA_LOG_STDOUT_H_ */

```

miehe-dup, 2019-08-23:
Oh, you're both confused and confusing. Brilliant :tipping_hand_woman:

AngerlFG, 2019-08-23:
I think I found it but still don't understand it :D
https://open62541.org/doc/0.1/group__logging.html#ga8cbe2af835c4e25bb9af7f3394031162
https://open62541.org/doc/0.1/group__types.html#gac6f19951777419d520d469d5130acae4

juangburgos, 2019-08-27:
Sadly before even being able to learn how to program in C, fist you need to learn about build systems (how to compile source code into applications). 

It is  a complex topic, and it varies whether if you are in Windows, Linux or any other platform. Go ahead and search for some "Beginner CMake Tutorials" on Youtube and once you understand what is going on, then go ahead and read [the documentation on building open62541](https://open62541.org/doc/current/building.html) on how to build the examples.

I personally recommend building the amalgamation, since you can avoid some missing dependencies errors like th eones you are having.

Sadly, nobody will help you with the basics in here, it is up to you to learn them using your own resources and the internet. Good luck!

AngerlFG, 2019-08-27:
I did the building. Now I'm trying to understand the functions. Unfortunately, I think the best I can do is to understand the examples since there are no functions definition in the documentation.

juangburgos, 2019-08-27:
If you want to use a simpler API and dont mind changing to Qt C++ maybe you can give a try to this library
https://github.com/juangburgos/QUaServer

andreasebner, 2025-02-04:
The question appears to have been answered. If there are new questions, feel free to create a new issue.

closed 2025-02-04 by andreasebner