open62541/open62541 #3037: Downloads from https://open62541.org/ do not currently contain open62541.c & open62541.h?

Jonas-Bartkowski, 2019-08-23:
When I currently download either the "Single-file distribution and full source code" or the "linux-64" release, they do **not** currently contain open62541.c and open62541.h, which I assume are the single-file distributions? I assumed I was just not finding it but "du -a open62541 | grep open62541\\." is not finding any .c or .h. Then I though "single-file-release" just means "single-file-release-after-you-built-the-single-file-release", but https://open62541.org/releases/ contains these files?

Sorry if this just ends up being an oversight from me.

ichrispa, 2019-08-23:
Hi @Jonas-Bartkowski 

if you clone and build master, the amalgamation script will run by default. The two files are created by joining all the existing sources into the two single-source files (.c&.h).

Have you run 
```
mkdir build
cmake ..
make
```
successfully after configuring the stack as you need it?

Beste regards,

ichrispa

Jonas-Bartkowski, 2019-08-25:
Hey @ichrispa 

Thank you for answering so quickly!

I did manage to successfully build it, I was just wondering why these files aren't included with the source-code, since the Download page is supposed to be there so you don't need to build the source-code, right?

But if you did this on purpose, you can close this -  I assumed you simply forgot to include them with the current release.

closed 2019-08-25 by jpfr