open62541/open62541 #3041: Connect to MX-AOPC UA Server

TheGringerEye, 2019-08-26:
how to connect to MX-AOPC UA Server(Moxa`s opc server for io logic) ? I am use qt5 and msvc 2017 compiler, open62541 in single-file release, more: https://pastebin.com/96h3eV2H

jpfr, 2019-08-26:
The log output tells the following story:
The SecureChannel is unencrypted. But you still have to encrypt the password as it is transmitted.

It seems your client is not configured for encryption.
See the docs and examples on how to do that.

Please reopen if required.

closed 2019-08-26 by jpfr

TheGringerEye, 2019-08-26:
But AOPC Server tuned on "none"
[image]

TheGringerEye, 2019-08-26:
I am dont undestend how to connect.