open62541/open62541 #3044: Weird variable typing

miehe-dup, 2019-08-26:
## Description

Using datasources, it is possible to create variables with values of mismatching type.

## Background Information / Reproduction Steps

Create server. Add a variable node, let its type be T1. Attach a datasource reading a value of type T2.

### Code example

This creates a variable "weirdThing" in the objects folder, having type `UA_Double` but a value of `(UA_Boolean) true`:

```c
static UA_StatusCode
readWeirdThing(UA_Server *server, const UA_NodeId *sessionId,
               void *sessionContext, const UA_NodeId *nodeId, void *nodeContext,
               UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
               UA_DataValue *dataValue) {
  UA_Boolean value = true;
  UA_Variant_setScalarCopy(&dataValue->value, &value,
                           &UA_TYPES[UA_TYPES_BOOLEAN]);

  dataValue->hasValue = true;
  return UA_STATUSCODE_GOOD;
}

// ... in main:

  // naming
  UA_VariableAttributes test_attr = UA_VariableAttributes_default;
  test_attr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;

  // creation
  UA_NodeId nodeid_test = UA_NODEID_NULL;
  UA_Server_addVariableNode(server, UA_NODEID_STRING(1, "weirdThing"),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
                            UA_QUALIFIEDNAME(1, "weirdThing"),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
                            test_attr, NULL, &nodeid_test);

  // data source
  UA_DataSource ds = {readWeirdThing, NULL};
  UA_Server_setVariableNode_dataSource(server, nodeid_test, ds);
```

Full Gist: https://gist.github.com/miehe-dup/8af913a11b80802c406a1efcf36f4a3c

### Used CMake options

(suspected irrelevant)

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): master: 07d989349137d6d6d5260c98528f651930d8de37
 - [X] Other OPC UA SDKs used (client or server): None
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-08-26:
We cannot prevent users of the SDK to build non-compliant servers in every case.

- The additional internal type checking means overhead for every read
- The receiving client should do a type-check of the received values anyway

To me this defaults to a WONTFIX.
At the most, we can add more type-checks that can be disabled in the config or via build flags.

Any more opinions?

miehe-dup, 2019-08-27:
> To me this defaults to a WONTFIX.

I fully second this. However, there is one point -- any other method of achieving this already is caught.

```c
  // vars
  UA_NodeId nodeid_test = UA_NODEID_NULL;
  UA_NodeId nodeid_test3 = UA_NODEID_NULL;
  UA_DataSource ds = {readWeirdThing, NULL};
  UA_Boolean value = true;

  // weirdThing (works!)
  UA_Server_addVariableNode(server, UA_NODEID_STRING(1, "weirdThing"),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
                            UA_QUALIFIEDNAME(1, "weirdThing"),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
                            test_attr, NULL, &nodeid_test);

  UA_Server_setVariableNode_dataSource(server, nodeid_test, ds);

  // creation weirdThing2 (fails!)
  UA_Server_addDataSourceVariableNode(
      server, UA_NODEID_STRING(1, "weirdThing2"),
      UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
      UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
      UA_QUALIFIEDNAME(1, "weirdThing2"),
      UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE), test_attr, ds, NULL, NULL);

  // creation weirdThing3
  UA_Server_addVariableNode(server, UA_NODEID_STRING(1, "weirdThing3"),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
                            UA_QUALIFIEDNAME(1, "weirdThing3"),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
                            test_attr, NULL, &nodeid_test3);
  // adding data fails!
  UA_Server_writeObjectProperty_scalar(server, nodeid_test3,
                                       UA_QUALIFIEDNAME(1, "weirdThing3"),
                                       &value, &UA_TYPES[UA_TYPES_BOOLEAN]);
```

This makes me feel like there is a working check already, just not in place for `UA_Server_setVariableNode_dataSource`?

jpfr, 2019-08-31:
The `write` service and the `addnodes` service are exposed to the network.
So we have to check for malicious/wrong input from a remote source.

The datasource is internal. Here we trust the local implementation of the datasource.
Otherwise we have to suffer the performance penalty of type-checking for every read.

closed 2019-09-02 by miehe-dup