open62541/open62541 #3048: Memory leak when UA_gethostname fails in addDiscoveryUrl

cabralfortiss, 2019-08-28:
In function https://github.com/open62541/open62541/blob/master/plugins/ua_config_default.c#L236
(I copy the code here for easier reading)
```C
static UA_StatusCode
addDiscoveryUrl(UA_ServerConfig *config, UA_UInt16 portNumber) {
    config->applicationDescription.discoveryUrlsSize = 1;
    UA_String *discurl = (UA_String *) UA_Array_new(1, &UA_TYPES[UA_TYPES_STRING]);
    char discoveryUrlBuffer[220];
    if (config->customHostname.length) {
        UA_snprintf(discoveryUrlBuffer, 220, "opc.tcp://%.*s:%d/",
                    (int)config->customHostname.length,
                    config->customHostname.data,
                    portNumber);
    } else {
        char hostnameBuffer[200];
        if(UA_gethostname(hostnameBuffer, 200) == 0) {
            UA_snprintf(discoveryUrlBuffer, 220, "opc.tcp://%s:%d/", hostnameBuffer, portNumber);
        } else {
            UA_LOG_ERROR(&config->logger, UA_LOGCATEGORY_NETWORK, "Could not get the hostname");
        }
    }
    discurl[0] = UA_String_fromChars(discoveryUrlBuffer);
    config->applicationDescription.discoveryUrls = discurl;
    return UA_STATUSCODE_GOOD;
}
```

Since discoveryUrlBuffer is not initialized, if config->customHostname.length is zero and later UA_gethostname fails (case coming from freeRTOS where this function is not implemented),  discurl[0] = UA_String_fromChars(discoveryUrlBuffer); will still be executed which will call strlen on discoveryUrlBuffer.

I see two options:
1. Return an error when UA_gethostname fails
2. Initialize discoveryUrlBuffer first character to '\0' so the discoveryURL becomes "opc.tcp://:4840/" although this doesn't look nice

And in parallalel, in the freeRTOS arch (and other arch that don't have UA_gethostname, instead of returning an error, return a fixed string which can later (or before) be changed anyhow, but won't produce an error or a funny discoveryURL like "opc.tcp://:4840/" in case the user forgets or is too lazy to change the hostname

cabralfortiss, 2019-08-28:
Also, I just realized that config->customHostname.length can never be truth because this function is always called after setDefaultConfig which sets everything to zero.

Pro, 2019-08-30:
We should do both. Initializing the variable reduces potential errors in the future. And if getHostname fails, then the discovery URL is invalid. This should result in an error return.

Is that the only place where addDiscoveryUrl is called?

cabralfortiss, 2019-08-30:
From a quick search, addDiscoveryUrl  is called from:

* UA_ServerConfig_setMinimalCustomBuffer
* UA_ServerConfig_setDefaultWithSecurityPolicies

I'll create a PR now to fix the leaking issue and the return value. It will still remain the problem that the customHostame is not setable so  if (config->customHostname.length) { is dead code

Pro, 2019-08-30:
Hm, there should be at least some unit test which checks that. Need to look at it in more detail

cabralfortiss, 2019-08-30:
Lines aren't covered:

https://codecov.io/gh/open62541/open62541/src/1.0/plugins/ua_config_default.c#L238

https://coveralls.io/builds/25438134/source?filename=plugins/ua_config_default.c#L238

Pro, 2019-08-30:
Ah, I knew that I implemented a test for it :+1: 
https://github.com/open62541/open62541/blob/1666968d9c6a4bdd7b5b7591dc804604e471d376/tests/server/check_server_userspace.c#L135

And the corresponding code which is tested there is called from here 
https://github.com/open62541/open62541/blob/1666968d9c6a4bdd7b5b7591dc804604e471d376/src/server/ua_server.c#L500

and inside of `ServerNetworkLayerTCP_start`

https://github.com/open62541/open62541/blob/1666968d9c6a4bdd7b5b7591dc804604e471d376/arch/network_tcp.c#L367-L386

This means that the network layer already use the correct discovery urls, but the `config->applicationDescription.discoveryUrls` is not correct.

cabralfortiss, 2019-08-30:
Nice! but now, there's the problem of this duplicated code. And there's still some unreachable code in addDiscoveryUrl. 

And now, with the PR, the setConfigMinimal will always fail for freeRTOS. Should we add a default hostname for the architecture?

Pro, 2019-08-30:
#3062 should fix all this?!

cabralfortiss, 2019-08-30:
Should work yes.

PR #3062 (2019-08-30): fix(server): Correctly initialize application description discovery urls for custom hostname
fixes #3048
commit a06a792f (2019-08-30): fix(server): Correctly initialize application description discovery urls for custom hostname

fixes #3048
closed 2019-09-02 by Pro