open62541/open62541 #3054: Error receiving the response with status code BadDecodingError

16akhila, 2019-08-29:
```
  Got this error while running sample client (available in open62541 examples) to dispaly all the nodes existed in the uaserver.
To breif my problem 
   I have successfully created  a uaserver  which can create 3000 /4000 nodes  and tested with  uaclient test application  where i can see all the 3000/4000 nodes 
Instead of  client application if i run open62541 sample client.c file it is able to display up to 3000 nodeids 
if i increase i to 4000 instead of showing nodeid it is throwing this error  

Here in sample client.c file 

  **UA_BrowseRequest bReq;
  UA_BrowseRequest_init(&bReq);
  bReq.requestedMaxReferencesPerNode = 0;
  bReq.nodesToBrowse = UA_BrowseDescription_new();
  bReq.nodesToBrowseSize = 1;

  UA_NodeId_copy(&nodeId, &bReq.nodesToBrowse[0].nodeId);
  bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL;
  UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);

  for ( i = 0; i < bResp.resultsSize; i++)** 
  Observed that  "bResp.resultsSize" value is 1 if 3000 parmaetrs on server side  &  value is 1 if 4000 parameters on server side.

Actual need is to detect 10000 parameters on client side while i am creating 10000 parameters in this open62541 library .

"Few details : Installed open62541 library git version on linux version 18.04 "

```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Git version
 - [ ] Other OPC UA SDKs used (client or server): Yes 
 - [ ] Operating system: Linux-18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

twaldowski, 2019-09-03:
Did you determine a resolution for this problem. I appear to be having the same problem when moving from 2000 to 3000 nodes. 2000 works with Integration Objects' OPC UA Client version 1.3.0, but 3000 files with the following errors:

[Error] |#| 2019/09/03 20:27:25 |#| Unexpected error processing response.
[Error] |#| 2019/09/03 20:28:23 |#| An error occured [sic] when browsing the node' s children. Exception : The operation has timed out.

16akhila, 2019-12-04:
4000 nodes works well on UAExpert UA_client  without any error . Just need to increase the time limit .
But with client.c code It doesnt even process the nodes above limit .
Any settings am i missing .

teccheck, 2025-02-13:
The CI pipeline covers this use case, so this should not be a problem anymore. Feel free to reopen if the problem persists

closed 2025-02-13 by teccheck