open62541/open62541 #3064: cannot include package open62541 v3.0

altafmahmud, 2019-08-30:
# I am trying to include open62541 into my sample project but failing with  fatal error: open62541.h: No such file or directory

## My build steps are:
mkdir build && cd build
cmake -G "MinGW Makefiles" -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=ON -DUA_LOGLEVEL=100 ..
nmake
nmake install

I see it's installed at: C:\Program Files (x86)\open62541
I am using Qt 4.6.1 as my IDE and also see the CMake Modules

From command prompt: g++ main.cpp
main.cpp:1:23: fatal error: open62541.h: No such file or directory

## My project's CMakeLists.txt:
cmake_minimum_required(VERSION 2.8)
project(Open62541_test)
add_executable(${PROJECT_NAME} "main.cpp")
find_package(open62541 REQUIRED)
if(OPEN62541_FOUND)
	message(fatal ${open62541})
	target_link_libraries(${PROJECT_NAME} ${OPEN62541_LIBRARIES})
endif()

Used CMake options:

```bash
cmake -G "MinGW Makefiles" -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=ON -DUA_LOGLEVEL=100 ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (0.3):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

[CMakeOutput.log](https://github.com/open62541/open62541/files/3560639/CMakeOutput.log)

Pro, 2019-09-04:
Can you show the content of the open62541 directory under C:/Program Files?

altafmahmud, 2019-09-04:
Directory of C:\Program Files (x86)\open62541

09/03/2019  08:57 AM    <DIR>          .
09/03/2019  08:57 AM    <DIR>          ..
09/03/2019  08:57 AM    <DIR>          include
09/03/2019  08:57 AM    <DIR>          lib
09/03/2019  08:56 AM         1,957,818 libopen62541.dll
09/03/2019  08:57 AM    <DIR>          share

Pro, 2019-09-04:
Can you recursively list the files in include subdir?

altafmahmud, 2019-09-04:
Here is the list below:

 Directory of C:\Program Files (x86)\open62541\include

09/03/2019  08:57 AM    <DIR>          .
09/03/2019  08:57 AM    <DIR>          ..
09/03/2019  08:57 AM    <DIR>          open62541
               0 File(s)              0 bytes

 Directory of C:\Program Files (x86)\open62541\include\open62541

09/03/2019  08:57 AM    <DIR>          .
09/03/2019  08:57 AM    <DIR>          ..
08/30/2019  12:35 PM               368 libc_time.h
08/30/2019  12:35 PM             8,291 ms_stdint.h
08/30/2019  12:35 PM             1,503 pcg_basic.h
08/30/2019  11:49 AM            24,073 queue.h
08/30/2019  12:35 PM             1,003 ua_accesscontrol_default.h
08/30/2019  12:35 PM            18,967 ua_client.h
08/30/2019  12:35 PM             4,706 ua_client_config.h
08/30/2019  12:35 PM            29,197 ua_client_highlevel.h
08/30/2019  12:35 PM             5,442 ua_client_internal.h
08/30/2019  12:35 PM            13,273 ua_client_subscriptions.h
09/03/2019  08:55 AM            10,073 ua_config.h
08/30/2019  12:35 PM             3,756 ua_config_default.h
08/30/2019  12:35 PM             3,592 ua_connection_internal.h
08/30/2019  12:35 PM           124,387 ua_constants.h
08/30/2019  12:35 PM               932 ua_log_socket_error.h
08/30/2019  12:35 PM               548 ua_log_stdout.h
09/03/2019  08:56 AM             1,454 ua_namespace0.h
08/30/2019  12:35 PM               757 ua_network_tcp.h
08/30/2019  12:35 PM               650 ua_nodestore_default.h
08/30/2019  12:35 PM             1,131 ua_pki_certificate.h
08/30/2019  12:35 PM             4,192 ua_plugin_access_control.h
08/30/2019  12:35 PM             3,863 ua_plugin_log.h
08/30/2019  12:35 PM             9,971 ua_plugin_network.h
08/30/2019  12:35 PM            20,469 ua_plugin_nodestore.h
08/30/2019  12:35 PM             2,098 ua_plugin_pki.h
08/30/2019  12:35 PM            18,315 ua_plugin_securitypolicy.h
08/30/2019  12:35 PM            11,901 ua_securechannel.h
08/30/2019  12:35 PM             2,592 ua_securechannel_manager.h
08/30/2019  12:35 PM             1,014 ua_securitypolicy_basic128rsa15.h
08/30/2019  12:35 PM             1,084 ua_securitypolicy_basic256sha256.h
08/30/2019  12:35 PM               755 ua_securitypolicy_none.h
08/30/2019  12:35 PM            50,470 ua_server.h
08/30/2019  12:35 PM             5,164 ua_server_config.h
08/30/2019  12:35 PM            14,383 ua_server_internal.h
08/30/2019  12:35 PM            20,746 ua_services.h
08/30/2019  12:35 PM             8,273 ua_session.h
08/30/2019  12:35 PM             2,092 ua_session_manager.h
08/30/2019  12:35 PM             6,542 ua_subscription.h
08/30/2019  12:35 PM             3,662 ua_timer.h
09/03/2019  08:55 AM             4,588 ua_transport_generated.h
09/03/2019  08:55 AM             7,023 ua_transport_generated_encoding_binary.h
09/03/2019  08:55 AM            11,602 ua_transport_generated_handling.h
08/30/2019  12:35 PM            35,425 ua_types.h
08/30/2019  12:35 PM             3,890 ua_types_encoding_binary.h
09/03/2019  08:56 AM            66,958 ua_types_generated.h
09/03/2019  08:55 AM           103,132 ua_types_generated_encoding_binary.h
09/03/2019  08:55 AM           164,110 ua_types_generated_handling.h
08/30/2019  12:35 PM             4,544 ua_util.h
              48 File(s)        842,961 bytes

Pro, 2019-09-04:
Now you should see the problem:
The include directory does not have a `open62541.h`. `open62541.h` is only generated if you use the amalgamated build setting, which is in general not recommended.
So you need to include the separate files from open62541. Have a look into the current example code in the git repo.

Also I would suggest to use version 1.0 instead of 0.3.

closed 2019-09-04 by Pro