open62541/open62541 #3068: version 1.0-rc3 instead 1.0-rc5

alekseikuleshov, 2019-09-02:
## Description

version 1.0-rc3 instead 1.0-rc5

## Background Information / Reproduction Steps
Hello 

This is my first question here. Please be patient. :)

I have cloned code from git repo, build and installed according documentation. The problem that I see version v1.0-rc3-112-g388c70f0 in open62541.h after compilation with option amalgamation enabled, but in git log on branch master I see commits from today 2 September 2019. I expected to see version rc5 at the moment.

I have tried couple times from scratch, but I'm still getting 1.0-rc3.

in open62541.h I see:
/**
 * open62541 Version
 * ----------------- */
#define UA_OPEN62541_VER_MAJOR 1
#define UA_OPEN62541_VER_MINOR 0
#define UA_OPEN62541_VER_PATCH 0
#define UA_OPEN62541_VER_LABEL "-rc3-112-g388c70f0" /* Release candidate label, etc. */
#define UA_OPEN62541_VER_COMMIT "v1.0-rc3-112-g388c70f0"

Can anybody help me to figure out why I'm not getting version 1.0-rc5?

I have played with cmake -DOPEN62541_VERSION=v1.0.3. and made v1.0.8 without errors. So I don't understand how it is work. :)

Thanks!

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): commit 388c70f0d723c64df8be61c52b880ed9e5d78671 
 - [ ] Other OPC UA SDKs used (client or server): server
 - [ ] Operating system: Ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-09-03:
The git repository contains different branches.
Switch to the "1.0" branch and try again.

Pro, 2019-09-03:
@alekseikuleshov was on master. The latest tag in the history of that commit was v1.0-rc3, therefore he got:

`v1.0-rc3-112-g388c70f0`

* `v1.0-rc3` is the latest tag for the commit
* `112` means there have been 112 commits after that tag
* `g388c70f0` is the commit id prefixed with a `g`

@jpfr I just pushed a new `v1.0-dev` tag based on master. Since 1.0 and master have diverged histories, this will not collide with the 1.0 branch. And on master all future version strings will start with `v1.0-dev` until we create a new tag based on master.

@alekseikuleshov please pull again with `git pull --tags` and your version should be updated. Maybe you need to delete the cmake cache before.

closed 2019-09-03 by Pro

alekseikuleshov, 2019-09-03:
I've got  v1.0-rc5-37-g395b254c on branch 1.0. Thank you!