open62541/open62541 #3082: Example 'server_nodeset_powerlink' is not working

ckmk14, 2019-09-05:
## Description
Using the master branch the example 'server_nodeset_powerlink' is not working:

```
info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: The value rank of ns=3;i=682 is incomatible with the value rank of the VariableType
info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Type-checking the variable node ns=3;i=682 failed with error code BadTypeMismatch
error/server	Adding the Powerlink namespace failed. Please check previous error output.
```

## Background Information / Reproduction Steps

Used CMake options:

```bash
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=ON
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=FULL

```

NoelGraf, 2021-11-05:
PR #4737

closed 2021-11-08 by NoelGraf

peschuster, 2024-07-22:
I think this is actually a bug in the Opc.Ua.POWERLINK.NodeSet2.xml and not in open62541.

The xml schema defines NMT_EPLVersion_U8 as ValueRank=1 array of bytes, but in the html documentation (and also the pdf download available to OPC UA members) it is shown as a single byte and not an array: https://reference.opcfoundation.org/POWERLINK/v100/docs/6.2

Also the powerlink to opc ua object mapping in Annex B.1 of the specification has NMT_EPLVersion_U8 as a single byte.